Ultimaker/Cura #19514: Relative Extrusion is ignored when importing G-Code

EmJay276, 2024-08-09:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

All

### Reproduction steps

- Slice any model with relative extrusion setting
- Import the G-Code

### Actual results

[image]

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[gcode.zip](https://github.com/user-attachments/files/16560211/gcode.zip)

smartavionics, 2024-08-09:
A quick look at the Cura 4 code shows that it only recognises relative extrusion (due to M83 in gcode) when the gcode flavour is RepRap.

GregValiant, 2024-08-09:
I see the same thing (looking in FlavorParser.py) and the behavior does appear to go back to at least 4.13.1.
Arc Welder's G2 and G3 lines continue to be unsupported as well.

EmJay276, 2024-08-09:
Should adding these lines to the "general" FlavorParser.py not just fix it?
https://github.com/Ultimaker/Cura/blob/e3287a1c161b63c75bc3dfd82257c0c5c2a69dca/plugins/GCodeReader/RepRapFlavorParser.py#L14-L20

EmJay276, 2024-08-09:
I just tested it in my local installation, and yes, it does fix it.
I will create a PR.

EmJay276, 2024-08-09:
You can test it in https://github.com/Ultimaker/Cura/pull/19515 if you want 👍

closed 2024-12-17 by HellAholic