Ultimaker/Cura #19517: Thingibrowser Add In Error: The API returned an error: Unknown.

sir-lancast3r, 2024-08-10:
### Cura Version

5.8

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

I try to open thingibrowser add in and it gives an error message "The API returned an error: Unknown." Happened after installing 5.8. 

### Actual results

"The API returned an error: Unknown."

### Expected results

The thingibrowser should open. 

### Add your .zip and screenshots here ⬇️

[cura5.8 Log.txt](https://github.com/user-attachments/files/16568120/cura5.8.Log.txt)

sir-lancast3r, 2024-08-10:
I've tried uninstalling the addon and reinstalling. 
I've tried uninstalling cura completely and reinstalling. 

Nothing seems to help. 
[image]
The drop down for my mini factory seems to work fine. 
[image]

Asterchades, 2024-08-10:
The Thingibrowser plugin is maintained separately to the main Cura project. It also hasn't been updated in over a year and a half - prior to Ultimaker's acquisition of Thingiverse, and well before Cura integration was added more directly to the website.

You'll probably want to open an issue over on its GitHub repository: https://github.com/ChrisTerBeke/CuraThingiBrowserPlugin