Ultimaker/Cura #19521: no start gcode

Scull215, 2024-08-10:
### Cura Version

5.7.1

### Operating System

windows 11

### Printer

elegoo neptune 4 plus

### Reproduction steps

i send the gcode and im not getting any part of the start gcode in the gcode file but i do have it in the slicer

### Actual results

no start gcode

### Expected results

start gcode then file

### Add your .zip and screenshots here ⬇️

[XYZ 20mm Calibration Cube - 1278865.zip](https://github.com/user-attachments/files/16570458/XYZ.20mm.Calibration.Cube.-.1278865.zip)

GregValiant, 2024-08-11:
Thanks for the report.
I've never heard of that before and I've never had the problem.
We will need a project file and the Cura.log file.  Zip them up and post them here.  The instructions are on the page you filled out for the report.

Scull215, 2024-08-11:
Can do when i can im away from the comp rate now
I tried to send the gcode instead of the stl but the gcode is unsupported

Sent from my Verizon, Samsung Galaxy smartphone

GregValiant, 2024-08-12:
You cannot post files here via Email.  You must come here and post the file.  All files must be "zipped".

Scull215, 2024-08-12:
[cura.log.zip](https://github.com/user-attachments/files/16587860/cura.log.zip)

LilBub, 2024-08-13:
One question - how is the gcode being sent? SD card? USB connection (i.e. Octoprint, etc.)  
I ask since I had a similar issue that required "Wait for Nozzle Heat" to be checked.

Scull215, 2024-08-13:
Fluidd to a neptune 4 plus
I also have a 3 plus with octoprint but I haven't tried that yet

GregValiant, 2024-08-13:
This doesn't seem to be a Cura problem.  Something you are doing locally (hardware, software, firmware) seems to be keeping the StartUp gcode from making it to the printer.  The log file indicates the StartUp is being added.

Have you tried saving to an SD card just to see if the gcode is correct when it comes out of Cura?

Scull215, 2024-08-13:
I had it working literally 2 nights before i entered in the bug? Haven't updated anything so how is that possible

Sent from my Verizon, Samsung Galaxy smartphone

Scull215, 2024-08-13:
I did save it to the desktop but it went straight to coordinates no preheat or purg line

Sent from my Verizon, Samsung Galaxy smartphone

GregValiant, 2024-08-13:
Do a slice, save the gcode file.  Zip the gcode file and post the zip folder here directly to the site.
This is the only report about this "Disappearing StartUp" problem and your log looked good.
It would be helpful if you carefully described your printing process from the time slicing ends until the printer starts to move.  Step by step.

Scull215, 2024-08-14:
I added 2 files 1 is resliced not even 10 min ago and the other I redownloaded from the printer the downloaded one was uploaded to it through the cura/moonraker connection with the start print button checked
[EN4P_xyzCalibration_cube1.zip](https://github.com/user-attachments/files/16607860/EN4P_xyzCalibration_cube1.zip)
[EN4P_xyzCalibration_cube (downloaded from printer).zip](https://github.com/user-attachments/files/16607862/EN4P_xyzCalibration_cube.downloaded.from.printer.zip)

GregValiant, 2024-08-14:
As you stated, there is no "StartUp Gcode" in either file you posted.
When I install an Elegoo Neptune 4 Plus in Cura the StartUp Gcode in the Cura Machine Settings is this:

;ELEGOO NEPTUNE 4 PLUS
M220 S100 ;Set the feed speed to 100%
M221 S100 ;Set the flow rate to 100%
M104 S140 ;Start heating extruder
M190 S{material_bed_temperature_layer_0} ;Wait for the bed to reach print temp
G90
G28 ;home
G1 Z10 F300
G1 X115 Y0 F6000
G1 Z0 F300
M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach print temp
G92 E0 ;Reset Extruder
G1 X115 Y0 Z0.4 F300 ;Move to start position
G1 X215 E30 F400 ;Draw the first line
G1 Z0.6 F120.0 ;Move to side a little
G1 X210 F3000
G92 E0 ;Reset Extruder

And that is what shows up in the Gcode file I generate.
If you go to "Settings / Printer / Manage Printers" and then click on "Machine Settings" what do you have in the StartUp Gcode box in the lower left of the dialog?
This is the stock installation of the Neptune 4 Plus
[image]

Scull215, 2024-08-14:
Is there some setting to not get it entered into the gcode that i dont see?

;ELEGOO NEPTUNE 4 PLUS
M220 S100 ;Set the feed speed to 100%
M221 S100 ;Set the flow rate to 100%
M104 S140 ;Start heating extruder
M190 S{material_bed_temperature_layer_0} ;Wait for the bed to reach print temp
G90
G28 ;home
M420 S1 Z10;Uncomment to enable progressive compensation height of 10mm
G1 Z10 F300
G1 X115 Y0 F6000
G1 Z0 F300
M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach print temp
G92 E0 ;Reset Extruder
G1 X115 Y0 Z0.4 F300 ;Move to start position
G1 X215 E30 F400 ;Draw the first line
G1 Z0.6 F120.0 ;Move to side a little
G1 X210 F3000
G92 E0 ;Reset Extruder

image.png (view on web)<https://github.com/user-attachments/assets/7720d72a-fac3-44ac-8d10-99c66f117a6c>

GregValiant, 2024-08-14:
I don't know.  There may be something in the MoonRaker plugin that is causing it.  As far as I can tell, if MoonRaker is not involved then the Gcode should contain the StartUp gcode as shown in Cura.  There are posts on Reddit regarding problems with the Neptune 4 Plus and MoonRaker.  5.7.1 is old enough that any incompatibilities with MoonRaker should have been discovered and fixed by the author.

Have you tried just copying the Gcode to an SD card and printing from the SD card?  Making everything as simple as you can would be a good place to start.  The Gcode file you save should have the StartUp Gcode in it.  (You can open any Gcode file in a text editor like Windows Notepad.)
As you add complexity to the workflow you will know when something breaks it.

Scull215, 2024-08-14:
I haven't yet there has been very little time I have been home and have been remote accessing for basically every so far come tomorrow morning I will be home for at least a few hours so I'll try then

Sent from my Verizon, Samsung Galaxy smartphone

Scull215, 2024-08-15:
[EN4P_xyzCalibration_cube.zip](https://github.com/user-attachments/files/16627163/EN4P_xyzCalibration_cube.zip)
Ya same thing happened, printer inbuilt homing at start of print then it tried to go straight to printing without preheating Klipper errored due to under minimum temp
note: yes with the start code it does normally home the printer twice at the start of every print

GregValiant, 2024-08-15:
When Cura slices something and puts together the toolpath into what will become the gcode file, the gcode file is stored in sections.
Section 1 is the opening paragraph from:
;FLAVOR:Marlin
to
;POSTPROCESSED

Section 2 is from:
;Generated with Cura_SteamEngine 5.8.0
to
;LAYER_COUNT:114

Section 2 contains more than just the startup gcode, and section 2 is missing in it's entirety.  That would seem to imply a badly behaving plugin, or post processor.

You have a non-Cura plugin installed.
**;Thumbnail generated by the Elegoo Neptune Thumbnails plugin version 4.4.0**
I think that plugin is behaving badly and deleting your "section 2" that includes the StartUp Gcode.  
So disable or un-install that plugin and try again.

GregValiant, 2024-08-15:
I'm going to remove the bug label and close this because the problem is that Elegoo plugin.
You can post a bug report on the the plugin author's github page [HERE](https://github.com/Molodos/ElegooNeptuneThumbnails/issues).

closed 2024-08-15 by GregValiant