Ultimaker/Cura #19523: Slicing fails in 5.7.1 and 5.8

danbi, 2024-08-11:
### Cura Version

5.7.1 and 5.8.0

### Operating System

MacOS Sonoma

### Printer

Voron 350 with 0.6mm nozzle

### Name abnormal settings

Tried all standard presets

### Describe model location

as loaded, sits on the build-late

### Describe your model

It works fine in Cura 5.6

### Add your .zip here ⬇️

[sand-spoon.stl.zip](https://github.com/user-attachments/files/16573739/sand-spoon.stl.zip)

GregValiant, 2024-08-11:
Thanks for the report.
I didn't have any trouble slicing it with my settings and a 0.6 nozzle.
You will need to attach your Cura.log file here and maybe a project file as well.
[image]

danbi, 2024-08-11:
It must be some corruption of the printer configurations then, as I was able to successfully slice for an Sunlu S9. Maybe also attach the printer configs too?

MrKrafter3D, 2024-08-12:
someone has the problem that the program crash when you´re programming the color chance with pause at height? Cura V5.8.0

danbi, 2024-08-12:
Here is Cura.log, from 5.8.0. Also attaching a project file with this model.

[cura-5.8-logs-profile.zip](https://github.com/user-attachments/files/16578298/cura-5.8-logs-profile.zip)

GregValiant, 2024-08-12:
If you are not using "USB Printing" then go to the MarketPlace, click on the "gear" icon, and scroll down and disable the "USB Printing" plugin.  If you have USB devices plugged in to the computer, and if those devices are configured as serial devices, then USB Printing will keep trying to connect even though it might be a pair of headphones rather than a printer.  That can cause problems and there are errors in the log that appear to be caused by it.

But there are a lot of other error descriptions in the log. 
You will need a UM person ( @HellAholic ?) to check that log file.  There are so many errors described in it that I get lost.  
The project itself sliced for me with no issues.

danbi, 2024-08-12:
> I took a quick look and have a guess as to why this has happened, did you by any chance did the install over the previous version?
> If so, (or just to be sure in general), remove the cura instances you have and install a fresh version (only 5.8.0 for example), it should still keep your printer definitions and your configuration folder intact.
>
> Looking at the logs it shows [error] Trying to retrieve setting with no value given: machine_extruder_start_code_duration which is pointing back to the definition of the printer, and looking a bit further up, the definitions seems to be affected in general.
>
> Give that a try and see if it works.

On this Mac I have three different versions of Cura (5.6.0, 5.7.1 and 5.8.0). Only 5.6.0 works. They have different working directories etc. This is Intel based Mac.

On another Mac I have much more: 4, 5.0, 5.1, 5.2.1, 5.3.1, 5.4, 5.5, 5.6, 5.8. All work. This is M1 based Mac.

It may be, that 5.7 did something wrong during upgrading the config directory which propagated further. I may try to completely remove any Cura remnants from the computer which will likely resolve it, but it is important to understand what the regression is. I vaguely remember there was some more radical change wrt Vorons in 5.7. The 5.7 was never installed on the second Mac.

Already did experiment adding new Voron2 350 printer, no improvement.

danbi, 2024-08-25:
I have experimented with this:

Disabled OctoPrint (3.7.3) for the printer, no help.
Deinstalled OctoPrint, no help.
Disabled USB Printing, no help.
Moved all thee instances of Cura from the Application folder, installed again Cura 5.8, no help.
Moved away the ~/Library/Application Support/cura directory, then started Cura 5.8.0 again and everything works.

So obviously there is some junk in that directory, most likely happened during the use of Cura 5.7.0 as any errors seen are since then.
Experimented by moving back the old cura directory, which had three settings directories for 5.6, 5.7 and 5.8, removed 5.7 and 5.8 but sill the same error. Which means there is something in my configuration, which does not bother Cura 5.6 (which works) but is not properly migrated to 5.7 or 5.8.

I'm attaching my ~/Library/Application Support/cura directory for reference.

[cura.zip](https://github.com/user-attachments/files/16739396/cura.zip)

danbi, 2024-09-07:
Eventually found the reason why this is happening. I had introduced new fdmextruder.def.json with the Sun S9 profile, which was apparently different from the one in Cura and in particular did not define machine_extruder_start_code_duration. The error about this was in cura.log. Instead of adding the definition from the Cura software I just deleted the file, which might have been outdated anyway, presumably the bundled one is used. 

Everything works so far, possibly Sunlu S9 should be investigated better.

Perhaps Cura 5.6 did not have/use this parameter and this is why it had no issues.

closed 2024-09-07 by danbi

GregValiant, 2024-09-07:
Nice bit of detective work.  I would not have found that in any reasonable amount of time (if ever).