Ultimaker/Cura #19526: Too heavy support

AirspeedHurricane, 2024-08-12:
### Is your feature request related to a problem?

Hi All,
You have done an excellent job making CURA, thank you. My only problem is that when I print fine objects, the support structure is often more substantial than the object, resulting in broken bits when I try to cut the support away. Could we have the option to make the support much lighter please?
Many thanks, Mike

### Describe the solution you'd like

This would prevent breakages on fine objects.

### Describe alternatives you've considered

I have considered resin instead, but don't really want more chemicals at home.

### Affected users and/or printers

Makers of intricate objects would benefit.

Asterchades, 2024-08-12:
Many options already exist to make the supports "lighter" than they default to. The most direct method is to reduce the **Support Infill Density**, but there are also the different **Support Pattern** options and the **Support Overhang Angle** to change how much of the object is given supports, plus you could also lower the **Support Flow** or **Support Line Width** to more directly compromise its structure. In all there's around 70 (more or less) different options which can be used to directly tweak how the supports are generated.

_Note that these extra options are only available in the Advanced/Custom view, which includes hundreds of other options above and beyond this as well. The Simple/Recommended view only offers a simple on/off checkbox._

AirspeedHurricane, 2024-08-12:
> Hi All,
Sorry, I found the setting I needed in machine profile

closed 2024-08-12 by AirspeedHurricane