Ultimaker/Cura #19527: Custom support mesh support distance priority not applying

WhyIcanBuyMe20atAnyMarket, 2024-08-12:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Sindoh 3DWOX — non-standandard configuration

### Reproduction steps

Add a custom support mesh to a model with a sloped supported surface
ensure that the support distance priority is set to 'z overrides x/y' (which is default)

### Actual results

The x/y distance overrides z at the overhang, ignoring the setting. This makes the overhang print terribly.

### Expected results

The z should override z/y as would happen with normal support

### Add your .zip and screenshots here ⬇️

[UCP_Fork_without_cutting_mesh.zip](https://github.com/user-attachments/files/16576317/UCP_Fork_without_cutting_mesh.zip)
With just the support mesh:
[image]
With normal support enabled in addition:
[image]
I don't just use the latter option as a work-around as where the default support drops down, it touches the part again, which makes the support difficult to remove.

WhyIcanBuyMe20atAnyMarket, 2024-08-12:
Incidentally, a similar issue with the bottom surfaces (of the support) in normal supports appears to have been fixed sometime between 5.7.2 and 5.8.0:
[image]
[image]

GregValiant, 2024-08-13:
Thanks for the report.
I can duplicate this and I think it's new in 5.8.0 as I don't recall this problem from previous versions.  

Your model is on the right and one of my calibration models is on the left.  They both show the same problem.  The angles are not being supported as they should be.  In this setup I have the Support Top Distance set to 0.0 so there should be no gap.  
[image]
That shape of mine has some flat areas that are correct with 0 gap, but they are parallel to the build plate so there really isn't an angle involved.

Although in the same slice, a flat "shelf" that is parallel to the build surface is correct with 0 air gap.
[image]

My calibration model has a round boss sticking out of one side.  You can see here that the supports don't follow the geometry as they should when Z overrides XY.  This is a model that I've sliced maybe 3000 times.  This is not what it's supposed to look like.
[image]

This is in Cura 5.7.1 and is the same setup showing that round boss.  This is how it's supposed to be.
[image]

Someone from the Cura team (@HellAholic?) will take a look.
This might be something that interests @ThomasRahm  as well.

ThomasRahm, 2024-08-14:
Did not yet debug it in depth, but based on what I am seeing I think HellAholic is correct here. It seems like `Minimum Support Roof Area` related. It does some fuzzy checks to combine overhangs on multiple layers before applying the `Minimum Support Roof Area` setting [Code doing the fuzzy check](https://github.com/Ultimaker/CuraEngine/blob/main/src/TreeSupportTipGenerator.cpp#L583C48-L606).
It checks the amount of interface layers up and down to combine these overhangs before applying `Minimum Support Roof Area` (assuming that they will combine to a single interface area).

@GregValiant If you increase the `Support Interface Thickness` does the behavior change for you? Also can you share the affected "Calibration Model" as a project file?

Regarding it being different in 5.7 to 5.8:
There was some fix to tips turning into roofs in https://github.com/Ultimaker/CuraEngine/commit/9cfea95a4ce3cd9942228c9990a8b1ee696ef02d which could influence the `fuzzy_area.removeSmallAreas(std::max(minimum_roof_area_, tip_roof_size_));` line in the fuzzy check. 
So I can imagine that it was violating the `Minimum Support Roof Area` in 5.7.

> In this setup I have the Support Top Distance set to 0.0 so there should be no gap.

You still have a `Minimum Support X/Y Distance` set which will still be enforced. (If it is 0 it will internally be set to 0.1 and then later the areas is offset again hoping to undo it being over-enforced) The support is not pushed down, but support too close to the model will still be removed. There is no perfect solution for this in my opinion, as there is no material that only has collision at the top, but not the sides.

GregValiant, 2024-08-14:
Hi guys.
I've changed the "Minimum Support Area" and Minimum Support Interface Area" down to 0.01 and 0.00 and achieved the same results.
"Support Interface Thickness" in the above images was 0.80.  The image below is with the Interface Thickness at 2.0 and both Minimum Areas at 0.01.
[image]

Changing the "Minimum XY distance" to 0.00 has some effect, but there is still a fair sized gap.
 
The angled surface has a horizontal hole in it and so you can see some support that appears to intersect the angled face, but it is actually supporting the roof of the hole.
The model is a support "torture test" I've been using for years.
[19527_GV_SupportShape.zip](https://github.com/user-attachments/files/16612079/19527_GV_SupportShape.zip)

ThomasRahm, 2024-08-15:
@GregValiant Sorry to be a bit annoying, but is is possible to also share the printer definitions so that I can import the project file?
[image]
Because with my settings there could be some issues if xy distance is 0 (Maybe relevant with soluble support material), but overall it seems fine.
[image]

GregValiant, 2024-08-15:
Sorry.  It's an Ender 3 Pro.  I open so many project files that I created my own definition file so my printer StartUp and Ending Gcodes don't get over-written.
[creality_greg3pro.def.zip](https://github.com/user-attachments/files/16624298/creality_greg3pro.def.zip)

ThomasRahm, 2024-08-16:
I just realized, this is about the interface behavior of regular support, not tree support. Please disregard everything I said above as it is not applicable to regular support.
Yes, the way regular support does Z overrides X/Y is something I think is technically wrong, causes no harm though.
If a overhang can't be supported, because it is not possible to place support below it (as it would be too close to the model) I think it should not be propagated downwards. Iirc this is the case though (and technically support too far down is better than no support at all). As such it kind of is Z overrides X/Y and X/Y overrides Z at the same time.

I do think something is broken here. If `Minimum Support X/Y Distance` is set to 0 there still is some gap.
[image]
If `Support X/Y Distance` is set to 0 that gap disappears:
[image]
Which by my understanding of that setting should not be the case.
(I have set interface wall line count to 1 here to better see which areas are supposed to be interface.)

P.S.

I am pretty sure though that this has lead me to an completely unrelated bug as this does not seem correct:
[image]

GregValiant, 2024-08-16:
[image]

That was with XY distance of 0.80, Min XY distance of 0.01, and Top distance of 0.20
I read the gcode for that slice into AutoCad to do some measurements.  In the above model, the angle on the right is 45° and the "Z" distance is 0.20 with an X distance of 0.40 (once I account for Line Width on a layer).
It's possible that this is just a trick of the preview.  I've never had to support an angle that low so I have no idea if this would show up in a print.  I'm thinking that it would not.  At a more reasonable (for me) support angle of 60° the distances won't matter near as much.  
It does look very odd though.  Maybe I can come up with a better test.  A 2D analysis probably isn't the best choice but it was quick and easy.