Ultimaker/Cura #19530: Printhead visualisation seems flipped in the Y-direction

rondlh, 2024-08-12:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

1. Set the Printer Printhead settings to see if the visualization is correct
[image]
2. Enable "Print Sequence" --> "One at a Time"

Check the print head visualization in the prepare view.
[image]

### Actual results

Set Y-parameters seem to be flipped in the visualization.
Y is the vertical axis.
The origin (0, 0) is at the bottom left.
So Y min (-5) is below the part, Y max (65) is above the part, but the shadow below the parts is much bigger than the shadow above the part.

Moving an object towards the left makes the object X position become smaller (go negative). Moving an object towards the bottom makes the object Y position became larger.
Again, the Y-direction seems to be flipped.

### Expected results

Printhead visualization correctly reflects the actual situation as defined.

### Add your .zip and screenshots here ⬇️

[Printhead Settings.zip](https://github.com/user-attachments/files/16590285/Printhead.Settings.zip)

GregValiant, 2024-08-12:
Thanks for the report.
I was just talking to someone else about the Cura coordinate system and it is truly confusing.

- For the purpose of Gcode and SOME things in Cura (like the Prime Tower location), the 0,0,0 is the left front corner for most cartesian printer (the mid-point of the bed on a few cartesian printers like Flash Forge and a couple of others) and all delta printers are "origin at center".
- For model placement the X0 Y0 is the midpoint of the bounding box that contains the model (the center of geometry), and the Z0 is the lowest point of the bounding box.
- The model coordinate Y axis is inverted so that +Y is toward the front of the build plate, but the main coordinate system has +Y toward the back of the build plate.
- For the "Print Head Settings" the origin point is the centerline of the nozzle.  -X (Xmin) is to the left, +X (Xmax) is to the right, -Y (Ymin) is to the rear, and +Y (Ymax) is to the front.
Are you still with me?

- Just to make things more wonderful, the "bounding box" of the model does not rotate with the model.  Instead, it changes size while still remaining square to the world.

Here is a 35mm cube.  Note the size in the "Scale" tool text boxes.
[image]

Now I rotate the model 45° and Prest-O-Change-O the translate tool now declares that the model is 35 /sin(45°) in both the X and Y.  Amazing.  Pythagoras would be dazzled.

If you have a couple of models on the build plate, and you "group" them, and then rotate the group, the "Bounding Box" will be visible.
[image]

I've beat my head against this wall before and I'll mark this as a duplicate.  You can check bug report #9363 and #2152 .
I wish you Good luck with this.

rondlh, 2024-08-12:
@GregValiant Thanks for the quick feedback and references to the related bug reports.

About the 35mm cube. Cura just reports the bounding box of an object, which would indeed be 49.5mm if you rotate the 35mm cube. This makes sense because if you can fit the bounding box in the build volume, then you can print the part. Tinkercad does the same.

For me the issue is not so much how Cura internally uses the coordinate systems.
I care about the visualization of the printhead. If this is incorrect, then printing "One at a Time" will cause issues, especially if you are have a printhead that is not symmetrical in the Y-direction.
Swapping the Y min and Y max values solves the issue for me.

GregValiant, 2024-08-13:
Swapping the Ymin and Ymax brings them into alignment with the "model" coordinate system with +Y being the front of the machine.  I have a 5015 fan hanging off the right side of my printhead, and a front intake duct on the hot end fan, so it's non-symmetrical in both the X and the Y.  That took a minute to figure out.

nallath, 2024-11-29:
Fixed for the next release. There is a preference that allows you to flip it!

closed 2024-11-29 by nallath