Ultimaker/Cura #19533: AquaSys 120 Material

Zhill18, 2024-08-13:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Ultimaker S7

### Reproduction steps

1. Updated Cura to 5.8.0
2. Synced materials in Ultimaker S7 printer
3. Notification received that AquaSys 120 material configuration is not available because it is not recognized.  
4. Went to Custom in material drop down
5. Selected Ultimaker Black ABS and AquaSys 120 maunally
6. Loaded a print with support material enabled 
7. Clicked Slice
8. Clicked Print to Cloud

### Actual results

Error message "The loaded material is incompatible with this print job. Please change the material before continuing." Regarding the AquaSys 120 material

### Expected results

Print starts and completes with no issues.

### Add your .zip and screenshots here ⬇️

[Cura Issue.zip](https://github.com/user-attachments/files/16601217/Cura.Issue.zip)

Zhill18, 2024-08-14:
> Hey @Zhill18 <https://github.com/Zhill18> ,
> I checked with the Digital Factory team regarding your issue. They have a
> request to check and rule few scenarios out, if you could provide some
> information regarding steps below. These are probably steps you already
> took but just to double check, please try them again.
>
>    1. Can you check in marketplace to see if you have the latest material
>    package installed?
>    2. Can you sync the material with the printer again through
>    preferences > configure cura > materials > sync with printers?
>
> Let us know if that changed anything.

Hello,

Yes, the latest material in the marketplace is downloaded. For AquaSys 120
that is version 1.2.0.

I also just performed the material sync again and it has all the current
material profiles. I tried to print again and still had the same issue.

Thank you,

Zhill18, 2024-08-14:
> Thanks for the check and additional information.
> One last request. Could you please check if the printer accepts the print
> file if you print via USB stick or through Network? (outside of Digital
> factory flow).

Gotcha, just confirmed the print file goes to the printer and then this
almost immediately pops up once it is sent.

[image: image.png]

I also tried to print via USB and this is what came up.

[image: image.png]
[image: image.png]
The material is loaded in bay F and in extruder 2.

Zhill18, 2024-08-15:
> The images did not load/show up, so unfortunately I can't see the result :(

Ahh okay.

So when I print via USB, I go through the proper steps and the
printer recognizes that there is a .ufp file on the USB and I can select
"Start Print." Once I do that an error message pops us that says "Load
material" & "Please load material ? ? into extruder 2." However the AquaSys
is already loaded in extruder 2.

When I print via network, a message pops up saying "The assigned printer,
NEUROSTAR-S7, required the following configuration change(s): Load unknown
unknown (unknown) as material 2 (This cannot be overridden).

Zhill18, 2024-08-15:
> Thanks.
> It appears there is a problem with the material package, we're following
> up with the manufacturer to resolve the issue ASAP.

Okay, thank you for the update!

Zhill18, 2024-08-26:
Hello!

Are there any updates from the manufacturer? We had to revert the Ultimaker
firmware back to 8.2.0, Stable version, and Cura software version 5.5 for
the AquaSys 120 material to work.

Zhill18, 2024-08-27:
> Aloha,
> they're updating the package, it takes a while since they have to resubmit
> the package and go through the process again.
> Should (hopefully) be an update next week.

Okay, perfect! Thank you!

Zhill18, 2024-09-03:
> @Zhill18 <https://github.com/Zhill18> We made a patch release to address
> an issue with the 3rd party materials not syncing, Cura 5.8.1 should
> resolve the not syncing issue.
> On the subject of the Aquasys, we're still waiting for new packages. I'll
> follow up again next week and keep you updated.

Awesome thank you for the update!

Flytking, 2025-04-27:
This is happening to me as well. Were there any fixes?

Zhill18, 2025-04-30:
@Flytking I had to manually add it as a material property/configuration and it started to work. There was no official update for the AquaSys package.