Ultimaker/Cura #19535: Slicing Failed after changing Moonraker Connection url

brianoliver1973, 2024-08-14:
### Cura Version

5.6.0

### Operating System

Windows 11 Home Build 22621.3880

### Printer

Ender-3 Max Neo BTT SKR Mini E3 V3

### Name abnormal settings

Chaned URL of Moonraker Connection

### Describe model location

Rotating 90 degrees slices, rotating back to original slices

### Describe your model

Yes, sliced succesfully before change

### Add your .zip here ⬇️

[CE3MAX_FreeSpa Aquarium Pump Impeller Cover V001.096.zip](https://github.com/user-attachments/files/16607056/CE3MAX_FreeSpa.Aquarium.Pump.Impeller.Cover.V001.096.zip)

GregValiant, 2024-08-14:
Thanks for the report.
Cura 5.5 and 5.6 had problems with certain shapes of models.  The problem was addressed and seems to be fixed starting with 5.7.0 and now into 5.8.0.  I suggest you upgrade.
I don't think the Moonraker connection has anything to do with it.
I'll mark this as a duplicate and go ahead and close it since the problem seems to have been fixed in the last 4 versions.

closed 2024-08-14 by GregValiant

brianoliver1973, 2024-08-15:
Already upgraded.

Thanks,
Brian