Ultimaker/Cura #19537: cura does not see the hole.

werewolftm, 2024-08-14:
### Cura Version

5.8.0

### Operating System

windows 10 x64

### Printer

prusa i3

### Reproduction steps

[image]
[image]

### Actual results

cura does not see the hole

### Expected results

cura does not see the hole

### Add your .zip and screenshots here ⬇️

[Power Bank - 18650x3 - 3661158.zip](https://github.com/user-attachments/files/16619074/Power.Bank.-.18650x3.-.3661158.zip)

GregValiant, 2024-08-14:
That is a sign that the model has errors.
Here you can see that the large model is red on top.  Cura colors surfaces that require support red and that surface has been colored because it is flipped inside out.
The model that is missing holes has the same sort of problem but it is the surfaces inside the hole that are flipped.  The surfaces are telling Cura that the inside is the outside.
The large model and the lower model were repaired using MS 3D Builder.  The one in the middle is the model you posted.  You can see that repairing the model fixes the problem.
[image]

I'll go ahead and remove the bug label and close this as the problem is in the models.  I hope you understand.

closed 2024-08-14 by GregValiant

werewolftm, 2024-08-14:
Thanks for the information. If it's not too much trouble, could you tell me what to press in MS 3D Builder to restore the model?

GregValiant, 2024-08-15:
Open the model in MS 3D Builder.
In the top left will be a button for "Import Model".  Click on that.
The model will automatically be checked for errors and if errors are found then a message comes up in the lower right informing you that the model needs repair (the model itself will have a red square around it).
Click on the message box to start the repair process.  I've only had a couple of models that were so bad they couldn't be repaired.  There are also free on-line repair sites if you want to go that way.

It's a good habit to check all models.  It can't hurt, and it can sure save some time.