Ultimaker/Cura #19538: TapStraal    Slicing went well in 5.7.2 butg in 5.8.0 this error occurred

brdi, 2024-08-15:
### Cura Version

5.8.0

### Operating System

OSX 11.7.10 (Big Sur)

### Printer

FLSUN V400

### Name abnormal settings

No

### Describe model location

On plate, does not slice after moving or rotating

### Describe your model

Yes in 5.7.2 with no problem

### Add your .zip here ⬇️

[FV_KDv2-Common.3mf.zip](https://github.com/user-attachments/files/16622722/FV_KDv2-Common.3mf.zip)

GregValiant, 2024-08-15:
Thanks for the report.
I didn't have any trouble slicing your project in 5.8.0.
Use the "Help | Show Configuration Folder" command to locate the Cura.log file.  Re-create the "won't slice" problem and then Zip the log file and post the zip folder here.

brdi, 2024-08-16:
Hi Greg,
Please find attached the log file
[cura.log.zip](https://github.com/user-attachments/files/16633870/cura.log.zip)
brdi

GregValiant, 2024-08-16:
Thank you.
I see this in there a few times and it seems to be the relevant part.

Starting export...
 [error] Trying to retrieve setting with no value given: speed_wall_0_roofing
 [info] Skipped 7 duplicate messages..
 [info] Closing connection.
ERROR - _onBackendQuit [1038]: Backend exited abnormally with return code 2!

That seems to mean that Cura couldn't find the "Top Surface Outer Wall Speed".  Looking into fdmprinter.def.json, if you don't over-ride that setting then it should be equal to the "Outer Wall Speed".

The Cura team will take a look.

brdi, 2024-08-16:
What I find in the content of the 5.8.0 app in fdmprinter.def.json is equal to the content in the 5.7.2 app:

  {
                                    "label": "Top Surface Outer Wall Speed",
                                    "description": "The speed at which the top surface outermost wall is printed.",
                                    "unit": "mm/s",
                                    "type": "float",
                                    "minimum_value": "0.1",
                                    "maximum_value": "math.sqrt(machine_max_feedrate_x ** 2 + machine_max_feedrate_y ** 2)",
                                    "maximum_value_warning": "150",
                                    "default_value": 30,
                                    "value": "speed_wall_0",
                                    "limit_to_extruder": "wall_0_extruder_nr",
                                    "settable_per_mesh": true
                                },

In both files the value is set to 30 as an absolute value.

BTW I've seen this behaviour before and reported a simular bug earlier this year.

brdi, 2024-08-16:
That was in #19111

GregValiant, 2024-08-16:
Thanks.
I just went over the other project file and it also sliced fine.  I jacked all the speeds up to 500 and there wasn't any problem.
Looking into the V400 definition file there isn't any mention of Speeds so whatever is in fdmprinter.def.json is the setting.

You will have to wait for a team member to go over this.  There may well be something in the log that I missed.

brdi, 2024-08-17:
OK, no problem 😎

brdi, 2025-01-12:
Same happend in 5.9.0. 
Is corrected. Issue closed.

closed 2025-01-12 by brdi