Ultimaker/Cura #19545: Cura 5.6 and 5.8: Incomprehensible print order and a lot off unnessesary travel compared to Cura 5.1.1

RabeSiebenstein, 2024-08-15:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

I recently updated Cura from an quite old version to v5.8.0 and now I am having issues with print order and travel time. Attached an example STL. 

To Reproduce: 
- Import STL and slice with Ender 3 v2, 0.4mm Nozzle, Standard Profile 0.2mm**

I tried various Cura versions and the issue seems to start somewhere between v5.1.1.  and v5.6.0.

5.1.1 (still good)
- Prints inner walls and shells in a optimized order (here pattern after pattern)
- Shortest travel that accounts for the enabled combing mode (one example travel highlighted in orange)
[image]

5.8.0 
- Prints all inner walls first and then all shells. Print head must go over all patterns twice.
- Cura adds a lot of unnessesary travel (one example travel highlighted in orange)
[image]

What I tried so far:
- Used Standard profile with 0.2mm layer height and without any custom settings
- Tried Cura 5.7.2, 5.6.0 (with same issue) and 5.1.1 (with no issue)
- Saved 3mf file in 5.1.1 and opened it in 5.8.0. Same issue after slicing

### Actual results

This increases printing time from 1h15min to 1h55min and prints fail because of the long travel distances and retractions.

### Expected results

Print order and travels should be optimized as in 5.1.1

### Add your .zip and screenshots here ⬇️

[ModelAndProjectFile.zip](https://github.com/user-attachments/files/16629889/ModelAndProjectFile.zip)

GregValiant, 2024-08-15:
Thanks for the report.
This is expected behavior because you have "Group Outer Walls" enabled which causes all the outer walls to print after all the inner-walls are complete.
Between 5.1.1 and 5.8.0 - 82 settings were added and 30 settings became obsolete.  The information is in the release notes of each version.
See if changing that setting makes a difference and let us know.  I'll remove the bug label for now.
[image]

RabeSiebenstein, 2024-08-16:
Thanks Greg, disabling "Group Outer Walls" fixes the print order issue. This experimental setting seems to be hidden and turned on by default, which is very confusing to people that just upgrade.

The issue with the travel is still present and I dont find any combing setting that improves it. Attached the 5.8.0 project file.

[AbflusssiebV4-Body580.zip](https://github.com/user-attachments/files/16640934/AbflusssiebV4-Body580.zip)

GregValiant, 2024-08-16:
Part of the problem is the model geometry.  I don't think the answer will be in "Combing".
You have a lot of holes to navigate around and you have "Outer-Wall", "Inner-Wall" and "Skins" to extrude.  That has Cura jumping around as none of the "skin" areas are actually connected to each other.  Remember, Cura (any slicer really) is designed for "most models" and not specific models.  Sometimes you need to fool the software.

I increased the "Line Width" to 0.45 (because of the distance between the holes) and set the wall count to "4" (to eliminate any skins that might be left).  I changed the Combing mode to "All" (because I felt like it).

Slicing your project it is a 1hr 8min print with 19% of that in Travel and **37%** in retractions.
Making those three changes it's only a 37minute print with 15% Travel and only **10%** in retractions.  That's a much more efficient print.

There is a knack to recognizing when model geometry plays a part in how Cura calculates the toolpath.  A model with Lots of holes is an indicator that you may need to come up with another solution and the solution will probably be different depending on if the holes are vertical - or horizontal.

RabeSiebenstein, 2024-08-19:
Hi Greg, thank you very much for looking into that and giving me such a detailed response.
I am taking away, that there were some improvements between 5.1.1 to 5.6. that fit most models, but created trouble in my particular case. Changing the "Wall Count" removed the weird travel (that was not present in 5.1.1) and even gained some efficiency by replacing the skin by additional walls.
I am closing the ticket because I no longer believe that this is a bug.

closed 2024-08-19 by RabeSiebenstein