Ultimaker/Cura #19546: Cura is skipping layers causing print failures.

rpweaverjr, 2024-08-16:
### Cura Version

5.5

### Operating System

Windows 10

### Printer

Ender 3 Pro, Upgraded, Marlin

### Reproduction steps

Built model in Fusion. 
Exported STL.
Loaded in Cura and generated gcode
Loaded on a thumb drive
Loaded on the printer

### Actual results

There are 5 items in the STL, 4 of these are printing fine but the main body prints to layer 2 and then not again until layer 12 causing print failure.
I've resliced this STL file 3 times and all 3 gcodes failed.

### Expected results

All the objects to print all layers.

### Add your .zip and screenshots here ⬇️

[BobbinPlate_1430 v2.zip](https://github.com/user-attachments/files/16631957/BobbinPlate_1430.v2.zip)

rpweaverjr, 2024-08-16:
I've been using Fusion and Cura for a long time. I've had a number of successful prints today and previous days using this same process.
[image]

rpweaverjr, 2024-08-16:
Previous versions of this file worked fine.

https://github.com/user-attachments/assets/895e7e57-44f9-4da8-8b7d-d55b981d6036

rpweaverjr, 2024-08-16:
Cura 5.0 Beta does not have this issue.

rpweaverjr, 2024-08-16:
I take it back about the 5.0. 5.0 prints the walls but the interior is still messed up. I'm beginning to suspect the STL.

Asterchades, 2024-08-16:
The walls in this model are only 0.2mm wide. These will not be resolved with the settings in the provided 3MF as you're using the default **Minimum Wall Line Width** of 0.34mm _without_ the use of **Print Thin Walls**. As such they're simply being ignored as they're too narrow. The 5.0 beta either incorporated a much narrower minimum feature size or has Print Thin Walls enabled by default.

I'm not sure that this model will ever print correctly without alteration, though. The video playback isn't working for me but this model is completely hollow - not the kind of "hollow" that would normally be subject to infill, but actual defined open space. Even with the walls being resolved you're still having un-supported islands starting out in open space (layer 10), and those holes are going to make a mess of any bridging when it finally reaches the upper surface (layer 12).

GregValiant, 2024-08-16:
I agree with Asterchades.  That model isn't really printable because it's a hollow box.  The base is 0.30 thick and the walls are 0.2 thick.  If I drop down to a 0.15 nozzle, the slice looks good but the print would fail unless there is support on the inside to hold up the large roof.
Here I've rotated the model 90° and dropped it into the build plate to show a cross-section of the body.  This is with a line width of 0.20.
[image]

This is the project and the only change is to enable "Generate Support".  Here you can see that the roof is supported, but at 0.40 line width, the walls won't slice because they are too thin.
[image]

At a 0.15 Line Width the slice looks as we would expect.  Without support the openings that have started would be printing over air.
[image]

I don't see a bug here.  The model slices fine given Cura settings that will work with that model.
Admittedly, those settings need to be ridiculously small to work, and the model would likely fail anyway.

I'll remove the bug label from this but I'll leave the report open so you can respond.  Let us know if you feel removing the label is premature.

rpweaverjr, 2024-08-16:
I was suspecting something like that but the previous version of the file printed a few days ago but not now. I didn't change anything in Cura. I typically only change bed adhesion (to brim for ABS), the infill amount,  75-100% and Supports, checked off and on. Whatever happened either happened in Fusion (I extended the plate and moved the curved area) or Cura. The unaltered STL of the previous version that printed 3 days ago will not print now.  If there were changes in Cura, they were accidental/inadvertent or made by Cura itself. I've even clicked Recommended to restore the original settings without any success.

Sent from my Verizon, Samsung Galaxy smartphone

image.png (view on web)<https://github.com/user-attachments/assets/70b952b7-5ec4-4902-9d6d-af6f5f2fdb13>

image.png (view on web)<https://github.com/user-attachments/assets/efda3087-4bf5-467d-b1b8-96edf925bcc5>

image.png (view on web)<https://github.com/user-attachments/assets/de4a74b0-4ca8-44e5-9227-43e0adcb9ad5>

rpweaverjr, 2024-08-16:
This might be a Fusion bug. One of the last things I did in the previous version of the file was to make the indentations larger with a "shell" function. I'm suspecting Fusion applied this to the entire plate rather than just the indentation. I'll see if I can't undo the shell by reverting to a previous version.
Thanks everyone.

rpweaverjr, 2024-08-16:
The previous version of the previous file doesn't have the indentations at all. I guess I'll try rolling back history in the current version of the previous file.

rpweaverjr, 2024-08-16:
Removing the "shell" step restored the volume of the plate. I will print it now.
Thanks

GregValiant, 2024-08-16:
I'll go ahead and close this then.

closed 2024-08-16 by GregValiant