Ultimaker/Cura #19548: Slicing issue

Rikolino24, 2024-08-16:
### Cura Version

5.8.0

### Operating System

Windows 11

### Printer

homemade 3d printer

### Describe your model

Vase shape

### Add your .zip here ⬇️

I use Cura with the 'spiralize outer contour' function to slice large circular parts in vase mode. However, during slicing, a 'ridge' forms on the part that I cannot eliminate. The error is barely noticeable when the layer height is less than 0.5 mm, but since I use a 3D printer to print concrete, I need a layer height of 10 mm. I would appreciate any help or advice to correct this issue. 

Thank you in advance!

[image]
[image]

GregValiant, 2024-08-16:
Post a project file and I'll look.  When I spiralize models I use @smartavionics fork of Cura which has some nice touches for Spiralize.  It's based on UM Cura 4.x.

Rikolino24, 2024-08-16:
Thanks man, I really appreciate it

(https://www.reddit.com/r/TalesFromThePizzaGuy/comments/7b2p38/thanks_man_i_really_appreciate_it/)
[Matera cónica 59capas.zip](https://github.com/user-attachments/files/16641969/Matera.conica.59capas.zip)

GregValiant, 2024-08-17:
I set up my printer with a 20.0 nozzle, and 10.0 layer height and a 20.0 Line Width.
This is in Cura MB 4.20.21.  You can see the jig -jog in the upper layers.
[image]

This is hard to see but it's the Gcode from that slice read into AutoCAD.  (I tried to catch the same viewing angle).
[image]

And this is the Gcode file opened in Cura MB
[image]

This is the Gcode file in the PrusaSlicer Gcode Viewer.
[image]

So I think it is a display glitch in the original preview because the jig-jog does not appear to show up in the Gcode file itself.
Are you seeing that aberration in the print?

smartavionics, 2024-08-17:
Hi. The Cura layer view is the main culprit here. It has never been able to display spiralized gcode well because it  cannot correctly display lines whose Z value differs from the Z value for the current layer. i.e. the displayed Z values cannot be between the layers so for a 10mm layer height Z, the Z values used for layer 2 will all be 20mm, whereas, the gcode will contain Z values that smoothly increase from 10mm to 20mm. Hope this makes sense.

Rikolino24, 2024-08-20:
I'd like to show you how the piece looks after printing. With the current settings, the "ridge" that appeared in the Cura preview is also visible in the printed piece. However, I noticed that GregValiant doesn't have the same issue, as his print path looks clean. I'll now try Cura Master, as you recommended. Thank you very much!

[image]
[image]

GregValiant, 2024-08-20:
I didn't find that jog in the gcodes I investigated.  It is certainly in the preview but it's the gcode that counts.

You can use the PrusaSlicer "Gcode Viewer" (which is a separately installed app) and it will take Cura out of the loop when viewing a gcode file.  It's what I used for the comparison shown above.