Ultimaker/Cura #19557: DLL load failed while importing pyArcus: 找不到指定的模块

runruiliu, 2024-08-19:
这个问题我已经解决 找到文件路径 pyArcus.pyd Arcus.dll  去安装个.exe 去路径里面把这两个文件 放到pyArcus.pyd  路径下 替换原文件  就可以工作了
_Originally posted by @srlihg66 in https://github.com/Ultimaker/Cura/issues/19192#issuecomment-2196347993_
Hello, when I performed the fifth step, I encountered the problem: DLL load failed while importing pyArcus: The specified module cannot be found. I have seen others mention this problem, but I don’t know the solution. Which .exe to execute, or is there any other way to solve this problem? Very much looking forward to your reply

runruiliu, 2024-08-19:
[image]
[image]

linjincan, 2024-11-21:
去官网下载一个Cura的exe  安装之后把   pyArcus.pyd Arcus.dll 拷贝到 编译的 venv\Lib\site-packages里覆盖