Ultimaker/Cura #19559: No way to avoid nozzle creating travel lines/scratches in top surface of flat print while traveling from one side to the other. No Combing or Z-hop options will prevent it.

just4747, 2024-08-19:
### Cura Version

5.8.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro w/ Marlin FW, BLTouch ABL, PEI plate

### Reproduction steps

I have tried all combinations of available Combing and Z-hop settings in Cura. The project file I attached has the following settings but I have tested others as well:

Combing Mode: All
Avoid Printed Parts When Traveling: On
Avoid Supports When Travelling: On
Z Hop When Retracted: On
Z Hop Only Over Printed Parts: Off

### Actual results

Looking at layer 19 for example (19 as listed in Cura at least), you can see that about halfway through the layer, the nozzle travels directly over the top of the surface printed earlier in the layer to travel back and forth between the Bee and Gun logos in the corners instead of combing over the walls or outside of the model. This happens no matter what combination of Combing and Z-hop options I try. I have asked in Reddit groups for help and no one is able to figure out why Cura is acting as if Combing is completely set to Off. I have also tried downgrading to a few earlier versions of Cura with the same outcome.

### Expected results

I expect the Combing and Z-hop settings to cause the nozzle to completely avoid taking the route of going right over the top of the print and making contact with the surface of the top layer at the time, which creates multiple scratch/trail marks that ruin the final print.

### Add your .zip and screenshots here ⬇️

[Cura Combing Issue Project.zip](https://github.com/user-attachments/files/16663318/Cura.Combing.Issue.Project.zip)
[image]
[image]
[image]
https://github.com/user-attachments/assets/7054ba4f-812b-479f-bd77-1ac5eab477d7

smartavionics, 2024-08-20:
Hi, my Cura 4 variant (available at https://github.com/smartavionics/Cura/releases) has recently added an option to use z-hop when crossing top-level skins. You may find it solves your problem.

GregValiant, 2024-08-20:
Thanks for the report.
I can duplicate this with your project and the current "Chep" profile.
If I change the "Wall Line Count" to 2 then the combing mode is respected.
When I switch to my own Ender 3 Pro the problem goes away and the combing mode is respected with the Wall Line Count = 1.
The problem also goes away with your printer if I re-select the "Chep" profile and discard all the changes _BUT_ the Wall Line Count must be > 1 or the problem persists.

What all that means is that I can't tell exactly where the problem lies.  There appears to be an issue with your "Chep" profile ("Ironing" is disabled but "Iron top layer only" is enabled), but the combing mode should work with a Wall Line Count of 1 as it does when my printer is active with my own setting profile.

Here the skin has finished and the nozzle is moving along the outer wall "combing" to work on the hornet.  This is with my E3Pro active, a single outer wall, and the draft shield is active.  This is how it should work.
[image]

Someone from the Cura team will take a look.  It's possible that the Wall Line Count = 1 is involved within Cura, but it might be something with the printer definition or your profile.  (I use a custom definition file because I open so many project files.  Consequently, although it's the same printer my definition file isn't the same as yours.)

just4747, 2024-08-20:
> Thanks for the report. I can duplicate this with your project and the current "Chep" profile. If I change the "Wall Line Count" to 2 then the combing mode is respected. When I switch to my own Ender [...]

Thanks for replying. Appreciate the testing. Unfortunately I am not really able to reproduce this solution, at least from what I can tell. I have tried a few things (again, I am checking all of this during layer 19 which is the layer _right BEFORE_ the top surface text is created (the Asian symbols and the A.K.A Jenny Vicious parts).

- I left the profile as-is and just changed the Wall Count to 2 or 3, and for both, I still see at least 1 (maybe 2) travel lines happen right across the top surface AFTER the top layer is put down, which I assume would still create a scratch line. Even though this may be less than the amount of lines before, I still do not understand why I am not seeing the nozzle go around the perimeter of the outer areas like combing should...
- I tried re-selecting the same CHEP profile it already uses and discarded all changes back to its default, and with Wall Count at 3, I do still see one travel line going over the top surface material, AND if I set it to 2 or 1, I again see _several_ travel lines. I just do not understand this.
- I tried selecting a non-CHEP profile (Cura Standard Quality .2mm profile) and see similar results to the above test - either 1 or several lines depending on Wall Count and Z Hop/Combing settings.

So it seems that no matter what I try with Wall Count, Combing, and Z Hop, I just cannot get this program to actually do combing properly at all. I have yet to see during a playback of this layer, the nozzle actually traveling outside of the print at all to cross to another section - not once. So frustrating.

GregValiant, 2024-08-20:
I've looked at it some more and I can't get it to work.  The model has a couple of errors (not unusual with lettering) and I repaired it in MS 3d Builder.  No love.
I rarely use "Monotonic" because it generates a lot of combing moves but on this model it does not.  I sliced one of my horseshoe shaped models that always involves a lot of combing moves and it looks like fine.
I sliced your model in 4.13.1, SmartAvionics 4.20.21, and PrusaSlicer and got the same funky travels across the skin.

My advice is to adjust your "Max Comb Distance with No Retract" to something small like 2.0.  Increase your Z-hop height to 0.5.  It isn't optimal but you may get it to print without marring that surface.

There is something about that design that the slicers just don't care for.  I translated the model from 3mf to STL and it still doesn't want to comb correctly.

just4747, 2024-08-21:
> I've looked at it some more and I can't get it to work. The model has a couple of errors (not unusual with lettering) and I repaired it in MS 3d Builder. No love. I rarely use "Monotonic" because it [...]

Hmm, so was your first success with fixing it just a fluke or did you ever get it to comb correctly earlier on?
All I did to create this model was to take an existing high res image in Photoshop, trace the surrounding lines to create a few separate layers/plates, exported each to PNG and convert to SVG using Inkscape, then positioned and merged each how I wanted to in Tinkercad. Same thing for the Asian symbols, but for the English text, I used Tinkercad's custom font builder tool (might be a user-created one, I forget) using an SVG of the font downloaded from a font site. None of that to me should result in model issues but I know none of what I did will come out perfect model-wise (though works for me 99% of the way otherwise). Not sure if any of this has to do with the issues, but I really hate that I can't have a proven way to make the stupid thing comb because this is far from the first time this has happened to me with a print (I just never looked deep into the Combing settings and tried them until now, so who knows about the past ones I had the same issues with...).

Thanks for looking into it though, if you happen to think of or find anything else, please me know. I will try what you said as well.

GregValiant, 2024-08-21:
I converted the STL to a DXF file and opened it in AutoCad.  There are edges that don't meet together.  Not open seams that would be a model error, just walls that "come close but don't touch" which leaves miniscule gaps between things.  I may scribble the model up in AutoCad and see how it goes.
And yes, I have not been able to duplicate the success I had at first.  That's one of the reasons I'm leaning towards it somehow being a geometry problem.  The fact that I get the same wonky results from 3 different slicers seems to lower the possibility of it being a software problem.

GregValiant, 2024-08-21:
I generated my own version of the model and I can't fix this.  I got it back to where with Wall Count at 2 it seems to work but when it's a single wall combing appears to be ignored in all slicers.
I tried with PrusaSlicer, OrcaSlicer, SmartAvionics Cura MB, and UM Cura 5.8.0 and they all show the same behavior on this model.
Here is my project for the dev's.
[19559 GV Combing.zip](https://github.com/user-attachments/files/16698840/19559.GV.Combing.zip)

just4747, 2024-08-22:
> Collaborator

Question - were these gaps only where the two separate top rectangular boxes meet the main model, which I am only using as modifiers for the top text layers? Or is it something unrelated to those two top boxes? 
Any screen shots or more info might help, in the case that it's something simple I can modify. Also, why doesn't Cura warn of model issues in this case then?

GregValiant, 2024-08-22:
Model errors are things like flipped normals or open seams (non-manifold).  There are some that Cura can catch and others that are only apparent in the slice preview (internal extra surfaces would be an example of that).
If the model is free of those sorts of "mechanical errors" in the mesh, then what's left must considered "design intent".  You can design a feature with a knife edge that simply won't slice because it narrows to a point, but that isn't a "model error".

In this case, and with either the original model, or my mock-up, there are no "errors" in the models.  We are stuck with a conundrum "why don't it work".
Here is a video snippet of the end of layer 19.  It's combing for a bit, but it's still cutting across the top skin.  I shuffle back and forth to highlight the erroneous moves.

https://github.com/user-attachments/assets/3b47cb86-123c-4dd4-8784-a1e228bf7e63

smartavionics, 2024-08-22:
I can't load Greg's project because I don't have the machine definition but just using the model with my own profile I see similar travels across the middle of the large zone unless I set the travel avoid distance to 5mm. Then I see this...

[image]

just4747, 2024-08-22:
> I can't load Greg's project because I don't have the machine definition but just using the model with my own profile I see similar travels across the middle of the large zone unless I set the travel avoid distance to 5mm. Then I see this...
> 
> ![Screenshot_2024-08-22_18-35-23](https://private-user-images.githubusercontent.com/585618/360571820-f313ad7c-4271-41ea-876e-bddb003338d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzNTEyMTQsIm5iZiI6MTcyNDM1MDkxNCwicGF0aCI6Ii81ODU2MTgvMzYwNTcxODIwLWYzMTNhZDdjLTQyNzEtNDFlYS04NzZlLWJkZGIwMDMzMzhkNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyMlQxODIxNTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MmM2NjgxMzg3NDJkYTUyNmRiMWIxMTUyYTFlNGY4NjQ5OWFmZDYwZjVlZDBiZWExNjU4YjZkN2Q4MDc2OWMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XfxSFaF7ymjFd9DC6rlbb8qbO8MT9YSSyyf1PMl8Kro)

This looks very different than  mine though in several ways - I see more walls and other changes that don't look familiar in yours. If I try to just change my Travel Avoid Distance to 5mm, it doesn't fix anything.

Can you load up the 3mf I originally posted and without changing anything but the mentioned distance setting, do you see the travel going across the surface during layer 19 as it starts printing the outer two shapes (the hornet and the gun on my version)?

Thanks.

smartavionics, 2024-08-22:
Hi, yes, I loaded your project and I get the bad lines until I set the avoid distance to 7mm and, crucially, turned off z hops! Don't know why that makes the difference and turning off z hops is probably detrimental elsewhere but there you go...

[image]

GregValiant, 2024-08-22:
That's the model I mocked up.  It's a "representation" and it's a good one as it doesn't work either.
I can get combing to work a little bit when I up the travel avoid distance as Mark suggested, but that isn't good enough.

https://github.com/user-attachments/assets/04d9550c-3db9-4362-942d-4ab817c1da91

smartavionics, 2024-08-22:
Incidently, I have always been reluctant to use z hops on machines with leadscrews driving the z axis as I imagine it will cause a lot of wear. I do use z hops on my Kossel mini delta as that has no leadscrews.

smartavionics, 2024-08-23:
Looking at the Cura code, I think the need to turn off z hops to make the nozzle route around the outside is a bug. I will be removing that requirement in my own Cura build.

GregValiant, 2024-08-23:
I lubed the lead screw with "real" high-moly machine grease and printed an extensible cover that collapses as the Z rises.  The cover keeps the dust and debris off.  Wear has not been a problem.  

Z-hops have their place but I enable them when I must rather than as a matter of course.

I do believe this is a good time to unveil another of _Greg's Fabulous Post-Processors_.
This one is _"Z-hops for Travel Moves"_.
It does not require that Z-hops are enabled in Cura.
Here are the settings...
[image]
That setup will cause .50mm Z-hops to be added from the start of layer 18 to the end of 19 if a travel move is more than 10mm.  That covers the long move from the hornet to the gun.
G1 F1500 E568.03559
G0 F600 Z4.1                   ; Travel Hop
G0 F9000 X78.207 Y146.722
G0 X78.484 Y146.713
G0 F600 Z3.6                   ; Travel Hop
G1 F1500 E574.03559

Note that It is a slow running script as it has to check the cumulative length of all travels. Not bad on this particular model but definitely noticeable on tall models.
Let me know if there are issues.
[ZhopOnTravel.zip](https://github.com/user-attachments/files/16728167/ZhopOnTravel.zip)

RobWickham, 2024-10-08:
I may have a more simple repro case. Using TinkerCad, I have a model with letters laid out on the build plate, layer 1 for 1.5mm. Travel moves are walking right over the top of the letters, despite combing=all, and avoid printed parts when traveling=true.
[image]
[image]