Ultimaker/Cura #19562: Spiral Skirt

webkonto, 2024-08-20:
### Is your feature request related to a problem?

It seems to me that the very fist cm of every print after the skirt/brim are essential to print a successful first layer and thus the whole part. The function of the skirt is to "bring the system in the right flow", get the PID working in "flow" conditions and fill the nozzle with material, which was flow out before during heat-up and Z-calibration. The actual skirt function does several lines with a step (seam) at a certain place. From the view of thixoropy and temp control, this "seam" stops flow completely (even possible to make a retract?) so the "nice flow" is gone, temp control (PID control) will react with a hard "D"-part reaction possibly. 

### Describe the solution you'd like

My proposal is to create a switch/checkbox to use spiral skirt (and brim if possible?) to support better thixotropy + temp behaviour and initial flow condition. The result should be a better material behavior at the beginning of the print process. Especially for thin first lines (outlines, holes, etc) at the first layer. It could be even nice to add a box like "do XX seconds of skirt minimum" to catch temperature behavior and PID control.

### Describe alternatives you've considered

Make skirt distance an large as possible, to reduce impact on "seams" in the skirt. But this is handwork for each part. Some designs do an "initial purge" but before printing the skirt, so this is more or less waste of material.

### Affected users and/or printers

Everybody, every printer design.

GregValiant, 2024-08-20:
It's an interesting concept.  I use purge lines (many printers include them in the StartUp) and so I often limit the skirt to a single loop for largish models.  I have found that TPU requires a skirt in addition to the purge lines, but I have a bowden printer and TPU just doesn't print well on it.  I haven't had any issues with initial layer adhesion or an incomplete start to the print while the nozzle "normalizes".

The "side-step" of the skirt (at the seam) is much like the side-step between (for example} inner-walls.  It is a "Line Width" index in the toolpath.  It would be exceedingly rare to have a retraction at that point unless the settings were such that there was a retraction at every side-step.  Without a retraction/unretract, the time duration of the filament stoppage is on the order of 10-15 micro-seconds for a 0.4 index, and the material is oozing during that time.  So for that short time period, there is some "unpowered" flow.

There would still be starts and stops in the skirt as every line segment would still have a beginning and an end.
Maybe someone else will chip in, but give that the skirt shape is a function of the model shape, I see this as a fairly complex solution to a problem that doesn't really exist.

webkonto, 2024-08-20:
I agree! It only pays, if it can be done with small effort. The effect is small, especially if you print with decent printing speed.