Ultimaker/Cura #19564: Cura Crashes PC

AviationNerd6000, 2024-08-20:
### Cura Version

5.8.0 // 5.7.1

### Operating System

Windows 11

### Printer

Creality Ender-3 S1 Pro

### Reproduction steps

1. Put model in
2. Pushed slice

### Actual results

Cura crashes my pc, sometimes it just goes into a black screen and restarts, sometimes the mouse stops moving and disappears and sometimes it has a blue screen

### Expected results

It should slice

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/16683107/cura.log)

Asterchades, 2024-08-21:
According to the log you're using a Raptor Lake S CPU (family 6, model 183) - either an i7-13700 or i7-14700 (or variant there-of) based on the thread count. As such it's entirely possible that you're being affected by the recent issues that are plaguing those CPUs, which could include instability up to and including crashes and restarts when subjected to load or even certain instructions (Cura does all of its slicing on the CPU currently).

I would encourage you to look further into these issues as it may simply be that Cura is not the only program you're using that's experiencing issues - it's just the one that's most obvious. There's a library in use for UE5 engine games which is known to be particularly problematic, so if you have access to any UE5 titles that may be a decent way to test.

https://en.wikipedia.org/wiki/Raptor_Lake#Instability_and_degradation_issue

LilBub, 2024-08-22:
I'd be shy to blame the CPU and curious if the computer has adequate RAM, storage space, decent video card, etc.
I have a laptop that just can't seem to run Cura either.  I'll check the specs of the processor.   I don't get a crash - it just won't load.

AviationNerd6000, 2024-08-23:
You would be correct, i have a intel i7-13700F, and yes some UE5 games have been crashing a lot, as well as unity, i will reach out to intel support to get my processor replaced.

closed 2024-08-23 by AviationNerd6000