Ultimaker/Cura #19566: after Merging models always z=1.6485mm

OWClocal, 2024-08-21:
### Cura Version

5.8.0

### Operating System

Linux Xubuntu 20.04

### Printer

Ultimaker S5

### Reproduction steps

After merging i always have either both models at z=1.6485 or sometime its only the "test" at z=1.6485
I can reproduce it every time.

### Actual results

The main problem is ofcourse if the "test" has z=1.6485 because its not touching the buildplate...that shows the last picture

### Expected results

that both files are merged and at Z=0

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[test-dual-print.zip](https://github.com/user-attachments/files/16690920/test-dual-print.zip)

OWClocal, 2024-08-21:
ok its definitive a Linux version bug, i did setup a VirtualBox VM with Windows 10 Pro and Cura 5.8 ....placed the two modells at the plate ...selected all and clicked merged ...and it did what i should do and not Z raising or any kind.

GregValiant, 2024-08-21:
Thanks for the report.
No matter how I orient them when they jump together they are at a non-zero Z.  I did manage to change it from 1.6485 to 6.6485 when they were rotated so they were standing up.
With "Automatically drop models to the build plate" checked in the Cura Preferences - the models do end up at "Z = 0".  There is also the right-click option to "Drop all models to build plate".
The Cura team will take a look.

[image]