Ultimaker/Cura #19568: [5.7.1] High amount of travel moves

ef301, 2024-08-21:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Creality cr10

### Reproduction steps

I sliced a boomerang Model, that contains shallow angles. on the top pattern the slicer planned long travels after individual short Lines.

- [ ] 

### Actual results

The printer took many unnessecariy and long travel moves after individual short lines, at those spots the print looks rough.

### Expected results

The printer takes many unnessecariy and long travel moves after individual short lines.

### Add your .zip and screenshots here ⬇️

[image]
[CCR10S_boomerang.zip](https://github.com/user-attachments/files/16700488/CCR10S_boomerang.zip)

GregValiant, 2024-08-22:
Thanks for the report.
Horseshoe and V-shaped models require more fooling around to keep the travel to a minimum.  There is going to be travel out to the extremities and that takes time.  I do have some suggestions.

Rotate the model 30° CW.
Set the "Top Bottom Line Directions" to [0, 90] so the skins are symmetrical on the print.
Set the "Infill Line Directions" to [0] (you are using grid so it only needs one number).  This will help the balance and strength.
Change the "Skin Expand Distance" to 1.2.  That will close up those little holes that form on layer 22.
Change the "Combing Mode" to "All" (there isn't much skin on this model).

That will drop the print time and travel will only be 14% of the total time (about 10 minutes less)
[image]

This is the way you have it set up:
[image]

You can try changing the Zseam location and the Layer Start X and Layer Start Y as well.

For a "V" / horseshoe shape, this isn't too bad.  The only way to really cut down on travel is to turn combing off, but then you will have artifacts on the print as the nozzle will travel between the two endpoints instead of taking the long way around.
I'm going to remove the bug label from this and close it as it is expected behavior given the geometry of the model.  I hope you understand.

Here is how I would set it up.  Your project showed 1:58 to print.  With those changes to the setup it's 1:45.
[image]

closed 2024-08-22 by GregValiant

ef301, 2024-08-22:
Thank you very much!