Ultimaker/Cura #19575: Different printing speed changes at different heights ?

RetArsonFD, 2024-08-23:
### Is your feature request related to a problem?

Is it possible to get a feature that would allow you to select different printing speeds at specific heights?  

### Describe the solution you'd like

On a bed slinger style printer, the taller the print gets, the more movement of the print is created as the bed moves.  If it was possible to slow the printing speed at specific heights, I think it might be able to achieve an overall better print quality. 

### Describe alternatives you've considered

I do not know what alternatives could be considered for this.  I'm sure somebody smarter than me might have an idea.

### Affected users and/or printers

I would think it would be beneficial to anyone with a bed slinger style printer.

GregValiant, 2024-08-23:
Go to "Extensions / Post Process / Modify gcode" and then "Add a Script".  Take a look at "Limit the X-Y Accel/Jerk...".
I designed it for this exact situation.  It isn't necessary to slow down but hard stops and starts can be tough to deal with.
Here are the settings...
[image]

That setup is for a gradual change.  Between layer 300 and layer 400 the Accel for the Y axis will gradually step down from 1000mm/sec² to 300mm/sec².  That will significantly soften the starts and stops of the bed.

The Change at Z post processor could also be used (it uses M220 to adjust the speed to a percentage of the set speed) and "Search and Replace" could also be used to change speed at a layer.
Search: =  ;LAYER:100
Replace: = ;LAYER:100\nM220 S75
Use Regular Expressions = no checked

Cura has a few ways to alter the speed at different heights this so I'll close this request.

closed 2024-08-23 by GregValiant

RetArsonFD, 2024-08-23:
THX !!!!!!!!   I had no idea that was there.  Now I just need to find something tall to print to run a test on it.  lol.