Ultimaker/Cura #19576: 1st layouts Slicing or printer Issue

Arteggg, 2024-08-23:
### Cura Version

5.8.0

### Operating System

windows 11

### Printer

ultimaker 3 extended

### Reproduction steps

I am not sure what happened but 1st layer is not matching with printed 1st layers. And+ it continue printing faulty. 
Is it slicing issue become or its printer faulty. I had print this type file before was ok. 

### Actual results

that is how is printing, I tried re sliced. restart printer still same
[image]

### Expected results

[image]
That is how it should look 1st lay out

### Add your .zip and screenshots here ⬇️

I am traying to add zip cura setting, but keeps saying we don't support this type of file.

GregValiant, 2024-08-23:
Thanks for the report.
That looks more like an adhesion problem than software problem.  It looks like the plastic isn't sticking and the nozzle continues to pull it around.  You might do better asking for advice on the [UM Forum](https://community.ultimaker.com/forum/106-ultimaker-3d-printers/).

If you use the "File | Save Project" command, and then in Windows Explorer right-click on the new 3mf file and then select "Send To", one of the options will be for a "Compressed (zip) folder".  There is a 25mb limit on file size that is imposed by GitHub but that would be the sort of file that will post here.

Arteggg, 2024-08-24:
Its sticking very well. 101% its definitely not this. I can see 1st layout moves and its draw as it prints but it wrong

sorry i was traying to add rar file
[cura setting.zip](https://github.com/user-attachments/files/16735667/cura.setting.zip)

GregValiant, 2024-08-24:
You have "Spiralize Outer Contour" enabled.  If the model is not specifically designed to be spiralized then what should be Travel Moves to "islands" are extrusions instead.  Looking at the print on the build surface that looks like what is going on, but looking at your posted image of the Cura preview that doesn't seem to be true.  I really need a project file to check to see if that is what's going on.

Arteggg, 2024-08-24:
Hi, 
I have tried to one off, spiralize. 
Its not that. 
Please find attached project file, also another picture form printer and from cure slicer 1st levels. Is not match. 
[image]
[P51D_fuselage2.zip](https://github.com/user-attachments/files/16737949/P51D_fuselage2.zip)
[image]

Arteggg, 2024-08-25:
[image]
Hi, 
I found the issue. it is Arc welding in special mode coursing this type of printing. 

Thank you GregValiant for trying to help

closed 2024-08-25 by Arteggg

GregValiant, 2024-08-25:
That sounds like it was something to do with the G2 and G3 commands that your printer firmware did not care for.
The 3mf file you attached is a simple model file.  A 3mf project file contains the model and all your Cura settings.  Without that it's hard to tell what might be going on.  I can't tell if you were in Surface Mode or using Spiralize.

If the incompatibility is with your firmware (G2/G3 not supported) there might not be anything that can be done.  If it is an incompatibility between Arc Welder and either Spiralize or Surface then maybe it should be looked at.
I'd still appreciate a proper "Project File" to be sure.