Ultimaker/Cura #19578: Custom printing temperatures do not update in the main UI

funkyrobot, 2024-08-23:
### Cura Version

5.8.0

### Operating System

windows 10

### Printer

Elegoo Neptune 2s

### Reproduction steps

If I'm creating a new material, for example PETG, but, I want to use a different Default Printing Temperature, i.e. 240 degrees. This does not get updated in the main UI.

### Actual results

What appears to happen when creating custom materials. In the Preferences dialog box, you are forced to use a standard naming convention i.e. PETG, PLA etc. Which has a specific temperature value predetermined and that is the temperature that displays in the main UI. which begs the question, why is Default Printing Temperature and Default Build Plate Temperature text fields even there?

### Expected results

When creating custom materials one should be allowed to use whatever temperature they see fit. 

### Add your .zip and screenshots here ⬇️

[Cura - Default Printing Temperture.zip](https://github.com/user-attachments/files/16733402/Cura.-.Default.Printing.Temperture.zip)

GregValiant, 2024-08-23:
Thanks for the report.
Try right-clicking on the input box and select "Use values from material".  See if that makes a change in the setting.
[image]

In the above case the temperature changed from the 215 to 205 which is what I have a custom PLA set to.

funkyrobot, 2024-08-24:
Hi Greg, thanks for the fast response, unfortunately, I do not have that
exact context menu. is there a plugging that will provide that option?

Phillip
[image: image.png]

fieldOfView, 2024-08-24:
It is part of the Material Settings plugin, which is available through the Marketplace dialog.

GregValiant, 2024-08-24:
There ya go @funkyrobot  - Words from the master.

funkyrobot, 2024-08-25:
Thanks, I'll check it out.

funkyrobot, 2024-08-25:
thanks, that worked

GregValiant, 2024-08-25:
Can I go ahead and close this?

funkyrobot, 2024-08-26:
Yes, go ahead. thanks again

closed 2024-08-26 by GregValiant