Ultimaker/Cura #19579: Model has weird holes at g-code.

Borab55, 2024-08-25:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Ultimaker 2 Extended (I'm using DIY printer, i'm using it for template)

### Reproduction steps

1-Just sliced my model.

### Actual results

There is some holes in the g-code.

### Expected results

Expected no weird holes at g-code.

### Add your .zip and screenshots here ⬇️

[Cura Bug Report.zip](https://github.com/user-attachments/files/16739430/Cura.Bug.Report.zip)
[image]
[image]

GregValiant, 2024-08-25:
Thanks for the report and thanks for the very complete file package.

I don't find any errors in the model.  When I change the "Wall Transitioning Threshold Angle" to 20° it slices completely.  Notice that you can see steps on the wedge surface.
[image]

I think this small model would be better suited to a smaller nozzle.  The 0.40 line width and 0.34 Minimum Line Width seem to be having an effect on that wedge shape as well as causing break-throughs at the vertical holes.
You also have some settings enabled that aren't need.  "Use Towers" and "Enable Bridge Settings" aren't required for this model and "Initial Layer Horizontal Expansion" isn't required when using a brim.

This is with a 0.2 nozzle and a 0.10 layer height and making the changes suggested above.  If I was to print this, I wouldn't bother with supports and 100% infill on this model doesn't provide any additional strength.  You would do better going with 3 walls.
[image]

I'm inclined to remove the bug label from this.  There are adjustments that can be mode to your settings that will allow this to slice as you would expect.  Let me know what you think.

Borab55, 2024-08-25:
If the problem is solved with what you said, it does not seem to be a bug as you say.  
Thank you for your suggestions, I will apply the settings you suggested in my free time.  It looks like I might try it with a 0.2 mm nozzle. Actually , I'm little afraid of it getting clogged.
 Thanks again.

closed 2024-08-25 by Borab55

GregValiant, 2024-08-25:
My Ender doesn't like the 0.2 nozzle unless I print really slow.  The diameter is so small that the flow rate is miniscule and trying to push more material through just doesn't work.