Ultimaker/Cura #19585: Slicing Failed with Interlocking Structure

trevorxo, 2024-08-26:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Snapmaker J1s

### Name abnormal settings

Generate Interlocking Structure

### Describe model location

Still no.

### Describe your model

Two rectangular prisms. I have sliced the models and printed them successfully without having them interlocking. When I uncheck that box, it slices just fine.

### Add your .zip here ⬇️

 I have tried updating to the newest Cura version, 5.7, I have tried 5.3, I have tried 5.5. I recently changed machines and this is the first time I tried to slice with this machine. Any model I try to slice will fail if Generate Interlocking Structure is checked.
[Dummy Files.zip](https://github.com/user-attachments/files/16752922/Dummy.Files.zip)

trevorxo, 2024-08-26:
[SJ1_Dummy Files.zip](https://github.com/user-attachments/files/16752939/SJ1_Dummy.Files.zip)
I only included the stl files in my original message. Here are the stl files and the 3mf project file.

GregValiant, 2024-08-27:
Thanks for the report.
I can reproduce this.

It appears that you have "Interlocking Depth" too deep.  When I change it from 10mm to 2mm it slices.  There seems to be an issue relating to the "Depth" and the thickness of the models.  It appears that you should have received a "won't slice" message rather than a crash.

The Cura team will take a look.

trevorxo, 2024-08-27:
Okay, the same thing works for me with these models. However, when I move one model to the edge of the other model so that they are not overlapping completely, then it is unable to slice. I used to be able to do that- have them partially overlap and still have an interlocking structure. If you can't recreate it, I can send the 3mf file.