Ultimaker/Cura #19586: Support Interface Grid Pattern bug

JParr91, 2024-08-26:
### Cura Version

5.8.0

### Operating System

Mas OS Sonoma 14.6.1 (23G93)

### Printer

Tronxy VEHO-600

### Reproduction steps

I applied the standard grid pattern on the support interface of my model with 0 walls set on the support interface.

### Actual results

There is an added concentric skin one layer thick that is created where its only touching the model on walls that begin after about 5 degrees of slope.

### Expected results

There should be no concentric elements on the grid pattern when interface walls are set to zero. This feature was not created in the support on version 5.2.0

### Add your .zip and screenshots here ⬇️

[Screenshots.zip](https://github.com/user-attachments/files/16753071/Screenshots.zip)

Asterchades, 2024-08-27:
It would appear that this is not strictly limited to the Grid pattern - Lines and Triangles are causing the same (mis?)behaviour. Understandably this increases the contact area in those locations beyond what was requested, sometimes significantly. Density also appears to have little, if any impact, as these pieces will form when set to 0%.

Regular support structures are unaffected by this - it is unique to the Interface.

ZigZag appears to show similar behaviour, but the nature of the fill type may well make it deliberate.

Someone from the team will take a look. I can't imagine this is deliberate given the disparity between the two different structure types.

Note in future that attaching images directly is preferable to hiding them in an archive.

GregValiant, 2024-08-27:
@ThomasRahm is this one you would be interested in looking at?

Asterchades, 2024-08-27:
It's model agnostic, so near as I can figure. I was able to reproduce it with an Akantor model I downloaded years ago, which is very different to the model seen in the OP's images save the presence of some areas of overhang.

[Overhang_Angle_Test_30-85.zip](https://github.com/user-attachments/files/16763737/Overhang_Angle_Test_30-85.zip)
[image]

I've grabbed [this](https://www.thingiverse.com/thing:1564848/files) off Thingiverse to replicate it for this project. Just with the default Custom FFF definition. Odd thing, though... you might notice it takes a long while to slice with the Interface set to 10mm thick like this - if you increase the Interface Thickness further you can increase that considerably.

ThomasRahm, 2024-08-27:
Thank you for the profile file.

Based on the comment at https://github.com/Ultimaker/CuraEngine/blob/1b78d4fe6ce1c4ccc1453c7a278eae209730c606/src/infill.cpp#L104 the behavior to generate walls if an area is too small is intentional.

I wonder why these are thin areas though, as they should be unioned with the larger interface area next to them.

I am pretty sure the issue was introduced with fractional interface. I strongly suspect the issue to be at https://github.com/Ultimaker/CuraEngine/blob/1b78d4fe6ce1c4ccc1453c7a278eae209730c606/src/support.cpp#L1737 and ensuring that this line is executed only if fractional roofs are present should fix this.

ThomasRahm, 2024-08-28:
Created a pull request. Note that with said pull request the here seen behavior only changes if the support top distance is a multiple of the layer height.
If there should be fractional support roofs, there is no good way to prevent this. It is either having the roofs that are a fraction of a layer height as a wall (because the area with the fractional roof is small) or discarding the fractional roof at angles that are so low that the fractional roof could not generate lines. I think the first option is preferable. If this is NOT the case let me know and I can change the pull request to instead discard small fractional roof areas.

rburema, 2024-10-23:
> I think the first option is preferable. If this is NOT the case let me know and I can change the pull request to instead discard small fractional roof areas.

I think the current proposed fix is the right solution.