Ultimaker/Cura #19587: 100% infill becomes skin and is unaffected by infill modifying mesh

Knot-Sure, 2024-08-26:
### Cura Version

5.8

### Operating System

windows 11 home

### Printer

Ender 3 S1 Pro

### Reproduction steps

Load a object with 100% infill
Load a mesh modifier for infill at 25%
change per part setting for overlap mesh infill density 25% and type triangles
merge and orientate 
slice

### Actual results

no infill is modified because it is all skin in the original 100% infill object slices as if no modifier was present

### Expected results

100% infill zone should be sliced as infill not skin
variable infill of 25% is selected zones

### Add your .zip and screenshots here ⬇️

[infill issue.zip](https://github.com/user-attachments/files/16753407/infill.issue.zip)
[image]
[image]

GregValiant, 2024-08-26:
Thanks for the report.
This has been mentioned previously.  I can't remember which version it was changed in, but the "expected behavior" now is to change to lines for 100% infill.  That may have been because some patterns leave gaps even at 100%.
The workaround is to set the main "Infill Density" to 99.9%.  That will keep it your previous infill pattern and the modifier mesh will work as expected.
The Cura team will take a look.
[image]

Knot-Sure, 2024-08-26:
Hi Greg!
Thanks for the update. Cura does in fact change to lines for 100% but upon
slicing becomes skin. I did the 99.9% and it works thank you!

-Brandon

Asterchades, 2024-08-27:
The default (going back further than is available on GitHub) behaviour is to actually remove infill from the process entirely when it's set to 100%. Rather than using Infill, as soon as you hit exactly 100% density it instead sets the **Bottom Layers** to 999,999 and **Top Layers** to 0 - effectively telling the whole model to generate as skin to make it solid unless it's one million layers tall or more. Since there's no longer any infill, infill modifiers no longer have any effect.

As with all such settings, however, it can be overridden. Definitions can change this behaviour, and the user can manually override it by changing the above values. Alternatively if the infill is defined using the **Infill Line Spacing** setting instead of the density it won't trigger the override. Indeed the Ultimaker definitions [changed this in 4.13.0](https://github.com/Ultimaker/Cura/commit/35550fc1b432806fd84c4d2d2354327373fd2da6) to behave as infill again, which in turn causes the resulting print to be faster (by default, skin only prints at half the speed of infill).

I'm actually not sure _why_ this is the default behaviour. There was obviously a reason 8+ years ago but whether it's still relevant is questionable, especially with the exception having been made.

GregValiant, 2024-08-27:
Not a workaround then, but changing a different setting.  I hadn't noticed the "999,999" bottom layers but there it is.
So @Knot-Sure,
99.9% infill or changing the "Bottom Layers" back to your preference will work.

From fdmprinter.def.json:
"bottom_layers":
"value": "**999999 if infill_sparse_density == 100** and not magic_spiralize else math.ceil(round(bottom_thickness / resolveOrValue('layer_height'), 4))"
and
"top_layers": 
"**0 if infill_sparse_density == 100** else math.ceil(round(top_thickness / resolveOrValue('layer_height'), 4))",

I've added overrides to my custom printer definition.