Ultimaker/Cura #19589: Cura slicer shows 190 layers.... klipper/fluidd says its 33250 layers

soundtechscott, 2024-08-27:
### Cura Version

5.8

### Operating System

win 10

### Printer

neptune 4  max - custom upgrades - klipper with octopus pro

### Reproduction steps

imported some objects into cura
sliced
sent to printer

### Actual results

Cura reports 190 layers
but klipper/fluidd says it 33250

### Expected results

it should be the same 190 layers in fluidd
works correctly on 5.7
so something with 5.8 slicing is causing it to be read as extra layers in klipper

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-08-27:
Thanks for the report.
You haven't given us anything to look at or review.  We need at least a project file.
You have (what appears to be) a heavily modified 3rd party printer.  When you "send to printer" how are you doing that?

soundtechscott, 2024-08-29:
My machine is a Neptune 4... but I have changed it out to an Octopus Pro Board instead of the piece of crap that Elegoo uses
everything else about it is Generic Klipper/Fluidd.. installed from kiauh, updated regularly thru the Fluidd interface

This File should be 33 layers according to Cura, but when you do a G-Code preview in Fluidd, it shows up as over 30k layers

I've Included 2 files that are different than the one pictures so I am not uploading files that I've spent a lot of time custom making
[image]

[image]

[GCODE end links.zip](https://github.com/user-attachments/files/16801592/GCODE.end.links.zip)

[CURA end links.zip](https://github.com/user-attachments/files/16801591/CURA.end.links.zip)

GregValiant, 2024-08-29:
Line 669 of the gcode file plainly states:
;LAYER_COUNT:33
If the softwares that are down stream are reading it wrong then the problem would seem to be downstream from Cura.

You appear to have the "Elegoo Neptune Thumbnails" post processor running.  That is provided with Elegoo Cura and not with UltiMaker Cura.  That might be causing a conflict downstream.

I'm going to remove the bug label from this and close it.  I just don't see an UltiMaker Cura problem here at all.  If you come up with something that actually points at UM Cura this can be re-opened.  I hope you understand.

closed 2024-08-29 by GregValiant

soundtechscott, 2024-08-29:
I am running Ultimaker Cura... with the elegoo thumbnail plugin added...
this is not elegoo cura version
I downloaded 5.8 directly from this github

tonight when I get home from work I can try it with the elegoo plugin removed

soundtechscott, 2024-08-30:
Removing the elegoo plugin didnt make any difference
I also submitted a bug report to Fluidd to see if its on their end

soundtechscott, 2024-08-30:
Cura used to work in fluidd as expected showing layers properly
now it doest...
This is a CURA bug

[image]

soundtechscott, 2024-08-30:
I'm going toad the recommended fix by the fluidd guys for now...
but It used to work just fine without it in earlier versions of cura

so something has changed or gotten dropped in cura 5.8(that existed in 5.7) and should be corrected to keep current with one of the more popular web/klipper interfaces for 3d printing

GregValiant, 2024-08-30:
Since someone had bothered to write a fix for this problem it would seem that this has been a known problem on the Klipper side.

Klipper is not currently one of the firmware flavors supported by Cura.  There is an ongoing project to add Klipper support into Cura, but it a work in process.
It may have worked for you previously, but since the configuration is after Cura has completed the gcode it would be up to the Klipper developers to handle the file that is received.
Your report is the first incidence of this behavior.  It would appear that most people are not having this problem as there is a good sized base of people successfully using Cura with Klipper enabled printers.

soundtechscott, 2024-08-30:
or since Cura is the only software that has this problem... its a problem on the CURA Side of things
my version 5.7 works fine, so I guess I'm staying in 5.7 for now
it works fine in orca, Prusa, and in Bambu slicers