Ultimaker/Cura #19595: Running from source issue while splash screen loading plugins

AbdElhalim12, 2024-08-29:
### Cura Version

5.9.1-alpha

### Operating System

windows 11

### Printer

N/A

### Reproduction steps

I cloned the repo
I followed the steps in the wiki 
I'm trying to run cura_app.py from powershell and vscode and pycharm and get that error everytime

### Actual results

PS C:\cura\Cura> & c:/cura/Cura/venv/Scripts/python.exe c:/cura/Cura/cura_app.py
[MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura
[MainThread] UM.Application.initialize [164]: App Version 5.9.0-alpha.0
[MainThread] UM.Application.initialize [165]: Api Version 8.8.0
[MainThread] UM.Application.initialize [166]: Build type None
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\share\uranium\plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\share\cura\plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\..\Resources\share\uranium\plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\..\Resources\share\cura\plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib\uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib64\uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib32\uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Resources\uranium\plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Resources\cura\plugins must be a folder.
[MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is C:\Users\abd_elhalem wahed\AppData\Roaming\cura\5.9
[MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is C:\Users\abd_elhalem wahed\AppData\Roaming\cura\5.9
[MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is C:\Users\abd_elhalem wahed\AppData\Local\cura\5.9\cache
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\.conan\91e8944\1\site-packages\plugins must be a folder.
[MainThread] UM.Qt.QtApplication.initialize [164]: Preferences file not found, ignore and use default language 'en_US'
[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\.conan\91e8944\1\resources\bundled_packages\uranium.json
[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: c:\cura\Cura\cura\..\resources\bundled_packages\cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...
C:\.conan\91e8944\1\site-packages\UM\Qt\QtApplication.py:701: DeprecationWarning: QEvent constructor is deprecated
  super().__init__(self.QtFunctionEvent)
[MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\.conan\91e8944\1\resources\bundled_packages\uranium.json
[MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from c:\cura\Cura\cura\..\resources\bundled_packages\cura.json
[MainThread] UM.PackageManager._loadManagementData [203]: User package management file C:\Users\abd_elhalem wahed\AppData\Roaming\cura\5.9\packages.json doesn't exist, do nothing
[MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [138]: Loading plugin configuration file 'C:\Users\abd_elhalem wahed\AppData\Roaming\cura\5.9\plugins.json'
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib\cura must be a folder.
2024-08-29 17:29:54,753 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura
2024-08-29 17:29:54,754 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.9.0-alpha.0
2024-08-29 17:29:54,755 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.8.0
2024-08-29 17:29:54,755 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None
2024-08-29 17:29:54,756 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\share\uranium\plugins must be a folder.
2024-08-29 17:29:54,756 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\share\cura\plugins must be a folder.
2024-08-29 17:29:54,757 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\..\Resources\share\uranium\plugins must be a folder.
2024-08-29 17:29:54,757 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\..\Resources\share\cura\plugins must be a folder.
2024-08-29 17:29:54,758 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib\uranium must be a folder.
2024-08-29 17:29:54,758 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib64\uranium must be a folder.
2024-08-29 17:29:54,759 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\lib32\uranium must be a folder.
2024-08-29 17:29:54,759 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Cura\venv\Scripts\plugins must be a folder.
2024-08-29 17:29:54,760 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Resources\uranium\plugins must be a folder.
2024-08-29 17:29:54,760 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\cura\Resources\cura\plugins must be a folder.
[... 416 lines cut ...]
  File "c:\cura\Cura\cura\UI\CuraSplashScreen.py", line 110 in showMessage
  File "C:\.conan\91e8944\1\site-packages\UM\Qt\QtApplication.py", line 589 in showSplashMessage
  File "C:\.conan\91e8944\1\site-packages\UM\Qt\QtApplication.py", line 213 in _displayLoadingPluginSplashMessage
  File "C:\.conan\91e8944\1\site-packages\UM\PluginRegistry.py", line 433 in loadPlugins
  File "c:\cura\Cura\cura\CuraApplication.py", line 861 in _loadPlugins
  File "C:\.conan\91e8944\1\site-packages\UM\Qt\QtApplication.py", line 235 in startSplashWindowPhase
  File "c:\cura\Cura\cura\CuraApplication.py", line 562 in startSplashWindowPhase
  File "C:\.conan\91e8944\1\site-packages\UM\Application.py", line 290 in run
  File "C:\.conan\91e8944\1\site-packages\UM\Qt\QtApplication.py", line 385 in run
  File "c:\cura\Cura\cura\CuraApplication.py", line 880 in run
  File "c:\cura\Cura\cura_app.py", line 243 in <module>
2024-08-29 17:29:56,324 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin CuraProfileWriter 1.0.1
2024-08-29 17:29:56,347 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin DigitalLibrary 1.1.0
2024-08-29 17:29:56,358 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin FirmwareUpdateChecker 1.0.1
2024-08-29 17:29:56,369 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin FirmwareUpdater 1.0.1
2024-08-29 17:29:56,379 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin GCodeGzReader 1.0.1
2024-08-29 17:29:56,386 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin GCodeGzWriter 1.0.1
2024-08-29 17:29:56,444 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-08-29 17:29:56.392] [info] Connected to 127.0.0.1:49674
2024-08-29 17:29:56,448 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin GCodeProfileReader 1.0.1
[... cut ...] der 1.0.1
2024-08-29 17:29:56,475 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin GCodeWriter 1.0.1
2024-08-29 17:29:56,489 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin ImageReader 1.0.1
2024-08-29 17:29:56,498 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin LegacyProfileReader 1.0.1
2024-08-29 17:29:58,942 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [540]: Loaded plugin MachineSettingsAction 1.0.1
PS C:\cura\Cura>

devgo3d, 2024-09-02:
I am also running into similar issue. Tried 5.8.0 and 5.8.1 same issue on local machine. 
https://github.com/Ultimaker/Cura/issues/19604 

Please let me know if you find some work around.

MeKKkKKkKK, 2025-08-28:
I am also got the same issue in 5.10.0, have u solved this problem?