Ultimaker/Cura #19596: Still cannot

cdsomerville, 2024-08-30:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Method X

### Reproduction steps

Tried to print using LABS extruder via CURA 5.8.1. 
No generic profiles available. I made a custom profile (I copied ABS-R just to work with ABS loaded in my Method X)
I am able to slice the file. 
I click "Print" to get an error citing Failed writing to specific cloud printer: MethodX not in remote clusters.

### Actual results

Cura does not recognize the LABS extruder as loaded.
I purged filament to no avail. 
I loaded and unloaded filament to no avail.
I am unable to use Cura to print via wireless network on the Method X

### Expected results

I would expect to be able to print on this machine. Still the best filament dryer I have. 

### Add your .zip and screenshots here ⬇️

[image]
[image]

Zezeinya, 2025-02-22:
same problem here. Where you able to resolve?

cdsomerville, 2025-02-22:
I avoid using this machine at almost all cost and have not verified if this issue has been resolved in months. I'll take another look, but it's doubtful.