Ultimaker/Cura #19597: STL file not opening, no STL reader plugin

Matee001, 2024-08-30:
### Cura Version

5.8.1

### Operating System

Ubuntu

### Printer

Anycubic Kobra Go

### Reproduction steps

1. Install Cura from flathub or website
2. Open Cura
3. Try to drag&drop stl file
4. Try to open stl file
5. Not working

### Actual results

The file doesn't open, doesn't appear on buildplate doesn't slice, doesn't have error message

### Expected results

File opens, appears on buildplate or if it doesn't it throws an error

### Add your .zip and screenshots here ⬇️

In the marketplace->plugins section there is no STL reader, only STL writer
cura.log file doesn't exist at the specified path
[image]

fieldOfView, 2024-08-30:
> 1. Install Cura from flathub or website

_Which_ website do you download Cura from? 

Ultimaker only makes and supports the builds downloaded from the Ultimaker.com website (which point to the releases here on github: https://github.com/ultimaker/cura/releases). That means only .appimage is supported on Linuxes. All other package formats are provided by third parties, and if those packagers omit certain files there is little that Ultimaker can do about it.

Matee001, 2024-08-30:
After about 5 minutes the file opened. There wasn't any loading bar or anything. The plugins still doesn't list STL reader.

fieldOfView, 2024-08-30:
Ah, you had me going with the focus on an STL reader plugin. Do not worry about that; I don;t know why the STL Reader is not listed; it is not listed for me on Windows either, but I can load STLs just fine.

Your problem seems to be that it takes very long for a file to open. That would be a duplicate of https://github.com/Ultimaker/Cura/issues/18968. In the Marketplace, try disabling the USB Printing plugin (and restart Cura).

closed 2024-08-30 by fieldOfView