Ultimaker/Cura #19602: Thumbnail icon not showing in elegoo neptune 4 Max

rajeshm5373, 2024-08-31:
### Cura Version

5.8.1

### Operating System

Linux

### Printer

elegoo neptune 4 Max

### Reproduction steps

I have used UltiMaker-Cura-5.7.1-linux-X64.AppImage and UltiMaker-Cura-5.8.1-linux-X64.AppImage. The thumbnail image is not showing on the printer screen. But, It shows the Elegoo icon

### Actual results

showing the thumbnail icon

### Expected results

need to show the slicer object 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-09-01:
Thanks for the report.
The links don't work for me.  I do not sign up to any accounts.
Elegoo has it's own format for adding thumbnails.  If the UM Cura post-processor "Create Thumbnail" doesn't work for you, you can try installing the Elegoo post processor.  That might work.

I'll remove the bug label and close this as it isn't a UM Cura problem.  I hope you understand.

closed 2024-09-01 by GregValiant

S117X, 2024-09-21:
Same issue here.

GregValiant, 2024-09-21:
Same response.  This is not an UltiMaker Cura problem.
Elegoo forked their own proprietary version of Cura and it's likely that their firmware is configured for their version.