Ultimaker/Cura #19606: Fan Speed NOT Changing During Print - Stays at Initial Speed

TheSchrund, 2024-09-01:
### Cura Version

5.8.1

### Operating System

MacOS

### Printer

Makerbot Method

### Reproduction steps

Fan speed and regular fan speed set to 100%
Initial fan speed set to 0%
Regular fan speed at layer 2

### Actual results

Fan speed is initially 0% and remains at 0% throughout the print

### Expected results

Fan speed should start at 0% for the specified number of layers then increase to regular speed of 100% at the determined fan speed.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-09-02:
Thanks for the report.
I can't duplicate this with my setup so we really need a project file.
Within the gcode file - Cura will put an M107 at the beginning of Layer:0 and then at the start of Layer:1 there should be an M106 S255 line.
You should see this near the beginning of the gcode file.
G1 F1500 E-6
;LAYER_COUNT:250
;LAYER:0
And then at the start of the second layer:
;TIME_ELAPSED:85.423495
;LAYER:1
M140 S50
M106 S255
And that is exactly what I get with all versions of Cura from 4.2 to 5.8.0 so this isn't likely a CUra bug but something that is happening locally with your setup.

There is a new post-processor "Advanced Cooling Fan Control" but if your printing process isn't getting the M106's through to the printer then it probably won't work either.

If your printer is running a flavor of RepRap firmware then it might need the "Fan scale 0 to 1" setting enabled.  That requires that the "Printer Settings" plugin is loaded from the MarketPlace.

TheSchrund, 2024-09-03:
Of course, the next time I opened Cura, it restarted like a new install, then said that my printer configuration was corrupt. It gave me the option to remove all my printers and remove all my profiles. I clicked yes to start fresh and now the fan commands are working properly. It will now start at 0% and  change to the selected % at the selected layer (with a step change increase in between).

Thanks for the prompt responses.

Now If i could just get the Method to listen Cura and switch extruders to allow multi color printing. If I select extruder #2 for one model it will pause after printing model 1 with extruder 1 then start printing model 2 with extruder 1.

GregValiant, 2024-09-03:
_"Now If i could just get the Method to listen Cura and switch extruders"_
That's a different problem (but maybe still firmware?) and really needs a different bug report.

TheSchrund, 2024-09-03:
The firmware should be able to handle it as SImplify3D can direct printing with either extruder. I have done multi extruder prints on the Makerbot Method using simplify 3D. I can assign each extruder to individual parts and the machine will switch. As for materials, I simply let the settings say PVA for second extruder and then just assign the correct temperature for the material i want to print with the extruder. Unfortunately I do not like the lack of other controls for some of the slicing so i may not keep simplify 3d and would like if the functionality can be added to cura.