Ultimaker/Cura #19607: [5.7.2] Adding cutting meshes results in missing layers

vitorio55, 2024-09-03:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Two Trees Bluer

### Reproduction steps

1. Import wood post model
2. Drag the wood post down so the base is excluded beneath the plate (see screenshot)
3. Add a cylindrical cutting mesh and configure according to screenshot
4. Replicate 2 more and drag the 3 cutting meshes to the top of the model
5. Slice

### Actual results

There are layers missing from the model in unexpected places (the wood post main body).
When I print the model it breaks during print because there are only some infill where some layers should be.

### Expected results

The cutting meshes at the top of the model should not impact the layers in other places.
If we import the wooden post model (also drag it down on step 2) and do not add the three cutting meshes at the top (skip reproduction steps 3 and 4) and slice it, all the layers are generated perfectly.

### Add your .zip and screenshots here ⬇️

[00WoodPost_inf75pp_HOLES_BUG_PETG.zip](https://github.com/user-attachments/files/16842176/
![cutting_mesh_holes_config](https://gith
[image]
ub.com/user-attachments/assets/61c93898-a253-4581-bec7-710c902a2fc5)
00WoodPost_inf75pp_HOLES_BUG_PETG.zip)
![exclusion_zone_plate](https://github.com/user-atta
[image]
chments/assets/62f604ee-82b4-4df6-8cdf-96aa2cd30948)
[image]

zawa999, 2024-09-03:
try this
https://bit.ly/4e4zLlq
select the desired compiler in the installer

GregValiant, 2024-09-03:
Thanks for the report.
The "missing layer" problem occurs 99.9% of the time because of an extra surface within the model that is fooling the slicer.  There are on-line model repair sites ( like [THIS ONE](https://formware.co/OnlineStlRepair) )  or you can get MS 3D Builder from the Microsoft store.  It's pretty good at repairing models and has decent tools for altering models.
If you post a project file I'll review it.  Without something to look at there isn't much we can do.

closed 2024-09-17 by github-actions[bot]