Ultimaker/Cura #19609: Tool Tips STILL getting in the way. No plugins installed

timgray, 2024-09-03:
### Cura Version

5.8.1

### Operating System

MacOS latest

### Printer

voron

### Reproduction steps

1. go manage printer settings
2.  edit that printer's settings
3. click on gcode in the tiny window to edit it
4.  become frustrate as that stupid help tip popup covers what you are trying to do

No I dont have any plugins installed,  this is the base help popus of cura and this problem has been here for 10 versions now.  PLEASE give us a way to disable that or make them smart enough to avoid covering up the control you are trying to use.
[image]

### Actual results

What happens is you get frustrated as you can not see what you are doing as the popup generator needs to have a disable feature in setup.

### Expected results

Expected results is that the tooltips avoid covering up what the user wants to work on.

### Add your .zip and screenshots here ⬇️

[image]

smartavionics, 2024-09-03:
Sorry, I don't have a solution but I notice that if I move the cursor into the field I wish to edit from the right hand edge, the evil popup is placed to the right of the field and doesn't get in the way.

GregValiant, 2024-09-03:
Yes there are workarounds BUT I'm going with @timgray on this one.  It's annoying.  Can it be attached to the "label" instead of the textbox?  Or maybe put on a 2 second timer?

GregValiant, 2024-09-04:
I'm a big fan of the way the StartUp and Ending Gcodes work now.  There can be a lot of trial and error getting the syntax correct but so far I've been able to get it to do what I want.
(Copying this from the StartUp box brought up the wonderful tooltip again.)

{'M92 E119' if 'TPU' in material_type else 'M92 E97'} ; Steps/mm for {material_type}
{'M206 X-116 Y-121 Z0' if machine_center_is_zero else 'M206 X-1 Y-6 Z0'} ; Home offsets {'(Origin at Center)' if machine_center_is_zero else '(Origin LF Corner)'}

So if I'm printing TPU the E-steps/mm change to accommodate it.  If I'm fooling around and have my printer configured as a "Origin at Center" machine, it's also accommodated.  Very cool.  Thanks for that.