Ultimaker/Cura #19612: no slicing take place in cura release 5.x

manfred-prt, 2024-09-03:
### Cura Version

5.8.1

### Operating System

windows 10

### Printer

Creality ender 5+

### Reproduction steps

Ich habe Files für ein 3D Flugmodell erworben und einige Teile davon lassen sich in Cura 5.8 nicht slicen. Rückfrage beim Hersteller ergab, dass es generell ein Problem mit Cura 5.x gibt. Nach seinen Angaben funktioniert das mit Cura 4.x . Dort werden aber einige aktuelle Parameter noch nicht unterstützt. In Anhang finden Sie eine Datei des Modells, die sich nicht slicen lässt.

### Actual results

Ich habe das File mit Cura 4.13.1 gesliced und da funktioniert es.

### Expected results

Mit Cura 5.x slice fehler
Mit Cura 4.13.1 slice ok.

### Add your .zip and screenshots here ⬇️

[Fuselage 2.zip](https://github.com/user-attachments/files/16852785/Fuselage.2.zip)

smartavionics, 2024-09-03:
Sorry, no, my Cura performs no better with this model. As you say, the  walls have zero thickness. Unprintable!

smartavionics, 2024-09-03:
Well spotted Greg. I've never used surface mode so it didn't occur to me to try it. I note that you can crank the line width up and get a chunkier print...

[image]

GregValiant, 2024-09-04:
Still, without any support the tops of the rectangular openings have to bridge a fair sized gap.  The first extrusions will sag until enough material builds up.
This is with mesh modifiers configured to print as support.  The model is at .8 line width and the support at .4 line width.  I don't know that I would trust this and the supports would be tough to remove.
A custom support structure designed specifically for this model would be a better alternative.  Cura just doesn't see anyplace that requires support so there is no "automatic" support structure.
[image]

A partial print like this could prove whether bridging the gaps would be doable, or if support would be required.
[image]

manfred-prt, 2024-09-04:
So i see that is impossible  to slice the modell im Cura 5.x (surface mode is recommendet and no support can be generated.
I will use Cura 4.13.1 to slice this modell and pass by the problem.

Thank you all for assistance .
I close this item for now.
Best regards Manfred

closed 2024-09-04 by manfred-prt

GregValiant, 2024-09-04:
Sorry we couldn't be more help.  Airplane parts and boat hulls are tough.  Some are designed to make use of "spiralize" while others (like this one) are comprised of just "surfaces" and they have no thickness.  The Cura variable line width generator apparently doesn't like that.