Ultimaker/Cura #19614: V400 temperature changes

Pippabear, 2024-09-04:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Flsun v400

### Reproduction steps

Choose esun petg black,  extruder should be 240 bed 80 changes when sliced. When I slice it keeps changing to extruder 267 bed 65.  Looking at the gcode I see this
M140 S60
M104 S267 T0
M190 S60
M109 S267 T0
Tried setting to generic petg and same thing happens.

### Actual results

When I slice it keeps changing to extruder 267 bed 65.  Looking at the gcode I see this
M140 S60
M104 S267 T0
M190 S60
M109 S267 T0

### Expected results

Gcode to set extruder to 240 bed to 80

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/user-attachments/files/16859402/cura.zip)

Asterchades, 2024-09-04:
According to the log provided, your Print Settings are such that the initial layer temperature *is* 267 and the bed *is* 60. The bed then appears to be set to remain there, while the printing temperature is set to drop to 252 for the final layer.

The question remains as to why. If I were to guess (which I have to, due to the absence of a project file which would store such detail), I would suspect that you're adjusting the Material Properties to tell it the new temperature, but at some point have manually overridden this in the actual print settings (either by typing them in yourself or using a profile which has them altered). To that end you should be able to restore the expected functionality by finding the various Temperature settings, right clicking on them, and selecting "Use value from material" to, well, do exactly what it says. There may also be an "ƒx" indicator which should do much the same if you left click it.

The Print Settings (the drop-down on the right of the main UI) have the final say in all settings that are used. Printer Definitions and Material Properties can only offer default values.

GregValiant, 2024-09-04:
A note (because I've been caught by this previously):  
_"Use value from material"_ is included with the "Material Settings" plugin available in the MarketPlace.
The "fx" button will be there with or without the plugin installed.

GregValiant, 2024-09-12:
@Pippabear we need a project file to investigate this.

Pippabear, 2024-09-12:
My Apologies,  I did a complete wipe of my install and did a reinstall and
that seemed to have resolved it.

GregValiant, 2024-09-12:
Thanks for getting back.  I'll go ahead and close this.

closed 2024-09-12 by GregValiant