Ultimaker/Cura #19617: Default File Name

Nathaniel8869, 2024-09-05:
### Cura Version

5.7.1

### Operating System

macOS Sonoma 14.6.1

### Printer

Longer LK5 Pro

### Reproduction steps

I just want to know if there's a way to change the way files are automatically named

### Actual results

When I save a sliced .stl file it has a prefix of "LLK5PRO_" and I'm wondering if there's a way to remove/change that

### Expected results

I'd like it if I could just save the file and it would name it based on the .stl file I dragged and dropped into Cura which it does, just without the prefix: "LLK5PRO_"

### Add your .zip and screenshots here ⬇️

I don't know what this is or how to use GitHub honestly, but customer service sent me here

GregValiant, 2024-09-05:
Under "Preferences" and "Configure Cura" on the General tab is "Add machine prefix to job name"..  Uncheck the box.
I'll remove the bug label for now.  Let us know if that worked for you.

Nathaniel8869, 2024-09-05:
No freaking way!!! That was a hella fast response, you're awesome, thank
you so much!!!!

closed 2024-09-05 by GregValiant