Ultimaker/Cura #19618: select face to align to the build plate failed

alfathur, 2024-09-05:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

ender 3

### Reproduction steps

1. open and place 3mf file to the build plate
2. select body
3. activate select face to align to the build plate
4. after that, select flate face in the body

### Actual results

the flat face in the body can't align to the build plate. the body just makes random movements on the build plate

https://github.com/user-attachments/assets/b04a43be-79e3-4f7d-805b-234c57ecfe72

### Expected results

flate face in the body should align to the build plate face

### Add your .zip and screenshots here ⬇️

[CE3PRO_atas4.zip](https://github.com/user-attachments/files/16883751/CE3PRO_atas4.zip)

https://github.com/user-attachments/assets/7ab14476-da81-4db9-8214-61fc247d90f2

GregValiant, 2024-09-05:
Thanks for the report.
I can't duplicate the problem.  Whatever face I pick immediately aligns with the build plate and the Z goes to "0".
This is with 5.8.1 on a Win10 Pro laptop.

https://github.com/user-attachments/assets/bf6a3d73-e62b-40ce-b51b-cfe8ad3bee28

Someone from the Cura team will take a look.

alfathur, 2024-09-05:
Thanks for repling,
When using my computer (run on windows 11) this problem gone, but appear in my laptop (also run on windows 11). 
I had tried to use the previous version of Cura (from version 4), but it still didn’t work.

alfathur, 2024-09-06:
re-installing didn't help, I tried many times! I think the problem is my laptop lol

GregValiant, 2024-09-06:
Everything was fine in Cura but when I played with that model in MS 3D Builder I got an odd result.

https://github.com/user-attachments/assets/418135ce-b32c-4b3a-a521-76c9a2e0e111

I think MS 3D Builder is taking the Center of Gravity into account.  The model would sit on that face if it was opened that way, but I could not get it to "settle" on that face on purpose.  It always fell over.

skiddyfisk, 2024-12-29:
I'm also experiencing this issue on a brand new install of 5.7.2 on my new PC. It SEEMS like it happens on parts with more complex geometry and not ALL parts - but the parts in question DO have flat faces for the build plate designed into them that should be selectable, and I am definitely clicking on those faces when this happens.

I tried to share some images but this form doesn't seem to be entirely functional either.

skiddyfisk, 2024-12-29:
[image]

Here we go. This piece wasn't cooperating, so I tried just making a big fat round coin shape with a huge flat surface to hopefully brute force this software into behaving correctly. This attempt was not successful.