Ultimaker/Cura #19620: Slice Failed when Tree Support Selected

formy2000, 2024-09-05:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 4 Max

### Name abnormal settings

Errors when Tree Support selected

### Describe model location

Fails at any location or rotation

### Describe your model

Have sliced before in earlier Cura versions

### Add your .zip here ⬇️

[image]

GregValiant, 2024-09-05:
Thanks for the report.
Cura 5.5 and 5.6 had issues with certain models.  since 5.7.0 Cura is much better.  I really haven't had any issues with 5.8..
I'll mark this as a duplicate and close it as the problems seem to have been fixed in later versions.  I hope you understand.

closed 2024-09-05 by GregValiant

formy2000, 2024-09-05:
The cura version I am using is a forked version created by Elegoo for the
neptune 4 max. Does 5.8 support this printer configuration?

GregValiant, 2024-09-05:
UltiMaker Cura does support the printer but there is something funky about how Elegoo developed the definition files used in Elegoo Cura.  There have been a couple of reports here where owners tried to copy the files to UM Cura and it didn't work so well.  
Right now I'm drawing a blank regarding exactly what the problem was.  If I remember the specifics I'll update this.
One problem was the "Thumbnail" generator.  Somehow it was causing problems, but there was something else as well.