Ultimaker/Cura #19622: Cannot open

sapbotgit, 2024-09-06:
### Cura Version

5.8.1

### Operating System

Windows 10 22h2

### Printer

Anycubic Kobra Neo

### Reproduction steps

just install latest version

### Actual results

Error. Logs:

Error traceback
```
Traceback (most recent call last):
  File "cura_app.py", line 243, in <module>
  File "cura\CuraApplication.py", line 923, in run
    self._welcome_pages_model.initialize()
  File "cura\UI\WelcomePagesModel.py", line 230, in initialize
    self._initialize()
  File "cura\UI\WelcomePagesModel.py", line 255, in _initialize
    "page_url": self._getBuiltinWelcomePagePath("WelcomeContent.qml"),
  File "cura\UI\WelcomePagesModel.py", line 220, in _getBuiltinWelcomePagePath
    return QUrl.fromLocalFile(Resources.getPath(CuraApplication.ResourceTypes.QmlFiles,
  File "UM\Resources.py", line 98, in getPath
    raise FileNotFoundError("Could not find resource {0} in {1}".format(args, resource_type))
FileNotFoundError: Could not find resource ('WelcomePages\\WelcomeContent.qml',) in 129
```

Logs
```
Thread 0x00000888 (most recent call first):
  File "C:\PROGRA~1\ULTIMA~1.1\share\cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 81 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001ff4 (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000570 (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\PROGRA~1\ULTIMA~1.1\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001898 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001850 (most recent call first):
  File "C:\PROGRA~1\ULTIMA~1.1\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000012c0 (most recent call first):
  File "UM\Backend\Backend.py", line 209 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000000dc (most recent call first):
  File "UM\Backend\Backend.py", line 189 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001ed0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001f3c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001934 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001c40 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000e50 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000015c8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001d78 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001d7c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00001ab8 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 169 in exceptHook
```

### Expected results

Start of the program

### Add your .zip and screenshots here ⬇️

[image]
[image]

sapbotgit, 2024-09-20:
it works if you load the model.

lik15, 2024-11-29:
I have the same problem. Have you solved it

sapbotgit, 2025-01-11:
@lik15 No, just loading from model opening with double-click.

adamkerz, 2025-01-12:
I have the same problem, in 5.9.0, fresh install, never had Cura on this machine before.

All the rest of the above details are the same.

[image]

adamkerz, 2025-07-17:
Sigh, this was a problem in pytest a few years ago too, and there's a warning in venv still to this day - misunderstanding of "symlinks" in Window; mount points aren't symlinks. Junction points are a third method of doing something similar, but I don't use them myself, so don't know the details.