Ultimaker/Cura #19624: Slicing failed (has been doing this a lot)

FooltoWorld, 2024-09-06:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Anycubic kobra Max

### Describe model location

Center of platform

### Describe your model

Tried to fix the mesh with meshtools

### Add your .zip here ⬇️

[111 Melee.zip](https://github.com/user-attachments/files/16913675/111.Melee.zip)

GregValiant, 2024-09-07:
Thanks for the report.
I tried with both Tree supports and Normal supports and in both 5.8.0 and 5.8.1 and I can't duplicate this.  I'm using a laptop and Win10Pro.
The Cura team will take a look.  This might be something that is system specific as there have not been many "won't slice" reports since 5.7.0 was released.

This is in 5.8.1.
[image]

FooltoWorld, 2024-09-08:
Thanks for the reply. Ever since I updated to 5.8 I have been having this
issue. Even some files that sliced perfectly fine prior, now are giving me
the same error