Ultimaker/Cura #19626: trying to print support structure mid-air

beb6, 2024-09-07:
### Cura Version

5.8.1

### Operating System

Windows 10 x64

### Printer

Ender 3 v3 KE

### Reproduction steps

I exported my 3d model directly from fusion 360 into cura and then proceeded to slice it. 

### Actual results

I noticed that some of my support structure are mid air. Not only that but cura decided that to start the layer it first wants to print the support Structure, which you've probably guessed starts mid-air.
[image]
[image]

I then kind of had doubts with myself and started googling for similar situations. Sadly none really matched mine. I hope u can help me out somehow because i dont really know what to do and i am also quite new to 3d printing :)
 

### Expected results

The Support shouldn't print in air, it should have been connected to a support tower.

### Add your .zip and screenshots here ⬇️

[CE3V3KE_Body1.3mf.zip](https://github.com/user-attachments/files/16920018/CE3V3KE_Body1.3mf.zip)

GregValiant, 2024-09-08:
Thanks for the report.
Set "Support Horizontal Expansion" to 0.0.  That is the setting that is causing support to stick out of the holes and then "Use Towers" wants to support those little bits.
(Looking at the screenshot It appears that you have "Use Towers" enabled, but it is not enabled in the project file.  Either way - turn off "Use Towers".)

That will be a very tough model to remove the supports from.  I would "Enable Bridge Settings" and turn supports off, and cross my fingers for luck.  A quick test print of just the "Area of Interest" might be a good idea.
Here I have sunk the model into the bed and then placed a support blocker as a Cutting Mesh with 0 walls, 0 tops/bottoms, and Infill density also set to 0.  That will just leave that tough area so you could see how it goes.
[image]

I'm going to remove the bug label and close this as it appears to be a settings issue.  Let me know if you feel that is premature.

closed 2024-09-08 by GregValiant

beb6, 2024-09-08:
Thank you for your support man. Bridging worked well on this model. Thank you so much! 
Have a wonderful day @GregValiant