Ultimaker/Cura #19628: All engines can be switched off

elektromos, 2024-09-08:
### Is your feature request related to a problem?

To level the bed, I always have to use the Repetier-host program, because it has a function that can be used to turn off all the motors.

### Describe the solution you'd like

Please take a look at this one function in the Repetier-host program, what a smart solution, that in manual mode, I can turn off all the motors! To what extent could this one function be included in the Cura program, then I don't have to use the other program! I put the Z axis in Home mode, and then I turn off all the motors, and I can happily level the heated bed.
Thank you!

### Describe alternatives you've considered

I don't understand the question

### Affected users and/or printers

I think everybody will be happy!

GregValiant, 2024-09-08:
I don't understand this.  When the stepper motors disable they lose their location.  This is generally indicated on the LCD by flashing "XYZ" location letters.

This also seems to be very machine and firmware specific.  Can the Gcode command that Repetier is adding simply be added to the StartUp Gcode in your installation of Cura?  Disabling steppers is usually accomplished with M84 or M18.

Turning off the steppers after auto-homing is usually a bad idea.  There is no guarantee that the "next Z=0" will be the exact same point-in-space as the "last Z=0" due to inaccuracies in the end stop switch.  My Ender can be off by +/- 0.10mm.

elektromos, 2024-09-09:
Hi!
Thank you for your help!
The M18 code actually shuts down the engines.
I would very much like you to put a small button in the Cura program, in
the Control printer section, which, if I press it, will execute the M18
instruction! Actually, that's all I would ask from the developers of the
program! I don't think this would take a professional programmer more than
half an hour :)
The push button should be included in the program because I will forget the
M18 in 1 week.
Best regards: József

GregValiant ***@***.***> ezt írta (időpont: 2024. szept. 8.,
V, 12:39):

GregValiant, 2024-09-09:
It seems to me that it is unlikely that the UltiMaker company will pay their people to customize the software for a 3rd party printer or because a user has a poor memory.  If you want to pay for that button, I'm sure a deal could be struck.

There is work being done to add more logic capability to the StartUp Gcode.  You may be able to use that to include your M18 in future releases.