Ultimaker/Cura #19630: [5.8.1 + earlier] appimage fails to open Fedora 40 - Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers. -  qt.qpa.wayland: EGL not available

Andrew-F-in-Australia, 2024-09-08:
### Cura Version

5.8.1

### Operating System

Fedora 40 - KDE

### Printer

---appimage fails to start - 

### Reproduction steps

May be a partial duplicate of #19303, unsure.

Attempted to open 5.8.0 appimage from commandline as well as through file manager (Dolphin.)
This was working 3 weeks ago when I last ran the printers.

This time, crash occurs on load (see screenshot below)

Attempted the following:
1. blacklisted nouveau (which used to work well,) and installed the nvidia proprietary drivers and secure boot.
2. tried previous kernels through the grub menu
3. tried previous appimages also.

4. Each attempt was met with the same error.
Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.
and:
 ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
qt.qpa.wayland: EGL not available

attempted to run as root - now only opens in superuser/with elevated privileges, but will not allow signin. Also detects a cyclic dependency.

`horace@fedora:/mnt/Data/3D Printer$ sudo ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
[sudo] password for horace: 
Sorry, try again.
[sudo] password for horace: 
qt.accessibility.atspi: Error in contacting registry: "org.freedesktop.DBus.Error.NameHasNoOwner" "Could not activate remote peer 'org.a11y.atspi.Registry': unit failed"
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMaU3bksc/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaU3bksc/share/cura/resources/qml/Actions.qml"
Running Firefox as root in a regular user's session is not supported.  ($XAUTHORITY is /run/user/1000/xauth_MZsJxd which is owned by horace.)
`
Log file:

### Actual results

when executed either from command line or file manager, program hangs with a qt wayland/EGL issue as above.
when executed from sudo, cura **does** opens but does not allow signin as a standard user without elevated privileges.
'horace@fedora:/mnt/Data/3D Printer$ sudo ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
[sudo] password for horace: 
qt.accessibility.atspi: Error in contacting registry: "org.freedesktop.DBus.Error.NameHasNoOwner" "Could not activate remote peer 'org.a11y.atspi.Registry': unit failed"
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMaU3bksc/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaU3bksc/share/cura/resources/qml/Actions.qml"
Running Firefox as root in a regular user's session is not supported.  ($XAUTHORITY is /run/user/1000/xauth_MZsJxd which is owned by horace.)
'

### Expected results

appimage should run, but doesn't open.

### Add your .zip and screenshots here ⬇️

[image]
cura log does not record error, just the last successful run of 5.8.0 on 12 August.

`
horace@fedora:~/.local/share/cura/5.8$ mokutil --sb-state
SecureBoot enabled
horace@fedora:~/.local/share/cura/5.8$ sudo inxi -G
Graphics:
  Device-1: Intel CoffeeLake-H GT2 [UHD Graphics 630] driver: i915 v: kernel
  Device-2: NVIDIA TU106M [GeForce RTX 2060 Mobile] driver: nvidia
    v: 560.35.03
  Device-3: Bison HD Webcam driver: uvcvideo type: USB
  Device-4: Logitech Webcam Pro 9000 driver: snd-usb-audio,uvcvideo
    type: USB
  Display: server: X.Org v: 24.1.2 with: Xwayland v: 24.1.2 driver: X:
    loaded: modesetting unloaded: fbdev,vesa dri: iris
    gpu: i915,nvidia,nvidia-nvswitch resolution: 1: 1080x2560~60Hz
    2: 2560x1080~60Hz 3: 1920x1080~144Hz
  API: EGL v: 1.5 drivers: iris,nvidia,swrast
    platforms: gbm,x11,surfaceless,device
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: intel mesa v: 24.1.7
    renderer: Mesa Intel UHD Graphics 630 (CFL GT2)
  API: Vulkan v: 1.3.290 drivers: N/A surfaces: xcb,xlib
horace@fedora:~/.local/share/cura/5.8$ dnf list installed '*nvidia*'
Installed Packages
akmod-nvidia.x86_64                                                                                                 3:560.35.03-1.fc40                                                                            @rpmfusion-nonfree-updates
kmod-nvidia-6.10.7-200.fc40.x86_64.x86_64                                                                           3:560.35.03-1.fc40                                                                            @@commandline             
nvidia-gpu-firmware.noarch                                                                                          20240811-2.fc40                                                                               @updates                  
nvidia-modprobe.x86_64                                                                                              3:560.35.03-1.fc40                                                                            @rpmfusion-nonfree-updates
nvidia-persistenced.x86_64                                                                                          3:560.35.03-1.fc40                                                                            @rpmfusion-nonfree-updates
nvidia-settings.x86_64                                                                                              3:560.35.03-1.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia.x86_64                                                                                          3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-cuda.x86_64                                                                                     3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-cuda-libs.i686                                                                                  3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-cuda-libs.x86_64                                                                                3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-kmodsrc.x86_64                                                                                  3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-libs.i686                                                                                       3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-libs.x86_64                                                                                     3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
xorg-x11-drv-nvidia-power.x86_64                                                                                    3:560.35.03-3.fc40                                                                            @rpmfusion-nonfree-updates
horace@fedora:~/.local/share/cura/5.8$ uname -a
Linux fedora 6.10.7-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Aug 30 00:08:59 UTC 2024 x86_64 GNU/Linux
`

furtber, 2024-09-09:
Hi @Andrew-F-in-Australia - I could get appimage cura 5.8.1 running on F 40 with downgrading mesa like this:

dnf downgrade mesa-dri-drivers-24.0.5

I know that's not a solution, but at least you can use latest cura ...

phako, 2024-09-21:
Work-around is to force Cura to run in XWayland with QT_QPA_PLATFORM=xcb ./ultimaker_cura.appimage