Ultimaker/Cura #19633: Cura - Problem Open with file stl

luigiamorfini, 2024-09-09:
### Cura Version

5.8.1

### Operating System

Linux, Debian etc.. (Flatpak user)

### Printer

Anycubic kobra plus

### Reproduction steps

My problem using Linux mint from flatpak cura version 5.8.1. 

I'am italian . no speak english

https://www.youtube.com/watch?v=YUQsZqAOC-k

### Actual results

My problem using Linux mint from flatpak cura version 5.8.1. 

I'am italian . no speak english

https://www.youtube.com/watch?v=YUQsZqAOC-k

### Expected results

My problem using Linux mint from flatpak cura version 5.8.1. 

I'am italian . no speak english

https://www.youtube.com/watch?v=YUQsZqAOC-k

### Add your .zip and screenshots here ⬇️

https://www.youtube.com/watch?v=YUQsZqAOC-k

Primy-games, 2024-09-12:
I had the same problem with Ubuntu 24.04.1 LTS and CURA 5.8.1. 
After uninstalling the USB Printing package.
Thank you

luigiamorfini, 2024-09-20:
re-upoad. problem cura open stl file: with linux ubuntu based or linux ubuntu 24.04. disable plugin usb priting .
https://www.youtube.com/watch?v=27JBOpOjmiA